r-lidar/lidR #216: grid_density help file error

SebastianKasan, 2019-01-31:
Hello Jean-Romain, 

I tried to open the help menu of the grid_density function in 2.0. Nothing happens and a while later R says:
Error in fetch(key) : 
  lazy-load database 'C:/Users/admin/Documents/R/win-library/3.5/lidR/help/lidR.rdb' is corrupt

Any suggestions?
Many thanks
Sebastian

Jean-Romain, 2019-01-31:
Try to reinstall and if there error persists [google it](https://www.google.com/search?ei=WNVSXKGRIMW9ggfvgbjwBQ&q=Error+in+fetch%28key%29+%3Alazy-load+database+is+corrupt&oq=Error+in+fetch%28key%29+%3Alazy-load+database+is+corrupt&gs_l=psy-ab.3...0.0..13956...0.0..0.0.0.......0......gws-wiz.SNDJxQHifmg)

closed 2019-01-31 by Jean-Romain

SebastianKasan, 2019-01-31:
Thank you, that has fixed it. It did not come to my mind that this is not a lidR specific problem.. Sorry