r-lidar/lidR #217: Supporting eigenvalue related feature calculation within lidR

komazsofi, 2019-02-02:
Currently, eigenvalue based feature calculation is extremely slow in lidR. I would like to suggest to speed up this calculation. 

Example code:

```r
library(lidR)

ShapeMetrics = function(X,Y,Z)
{
  xyz=rbind(X,Y,Z) 
  cov_m=cov(xyz) 
  eigen_m=eigen(cov_m)
  
  shapemetrics = list(
    eigen_largest = eigen_m$values[1],
    eigen_medium = eigen_m$values[2],
    eigen_smallest = eigen_m$values[3],
    curvature = eigen_m$values[3]/(eigen_m$values[1]+eigen_m$values[2]+eigen_m$values[3]),
    linearity = (eigen_m$values[1]-eigen_m$values[2])/eigen_m$values[1],
    planarity = (eigen_m$values[2]-eigen_m$values[3])/eigen_m$values[1],
    sphericity = eigen_m$values[3]/eigen_m$values[1],
    anisotrophy = (eigen_m$values[1]-eigen_m$values[3])/eigen_m$values[1]
  )
  return(shapemetrics)
}

# Set working directory
workingdirectory=""

ctg <- catalog(workingdirectory)

opt_chunk_buffer(ctg) <- 1
opt_chunk_size(ctg) <- 500
opt_cores(ctg) <- 18

metrics = grid_metrics(ctg,  ShapeMetrics(X,Y,Z), res = 2.5)
plot(metrics)
```

Explanation of the calculation (from [Lucas et al, 2019](https://www.mdpi.com/2072-4292/11/3/292)):
The point cloud Ῥ is a set of points {p1, p2, ..., pn} \(∈R^3 \), where each point pi has x, y and z coordinates. describe the distribution of points of a neighborhood in space. We used the local structure tensor to estimate the surface normal and to define surface variation. The structure tensor describes the principal directions of the neighborhood of a point by determining the covariance matrix of the x, y and z coordinates of the set of neighborhood points and by computing and ranking the eigenvalues (λ1, λ2, λ3, where λ1 > λ2 > λ3) of this covariance matrix. Hence, the magnitude of the eigenvalues of the matrix describes the spread of points in the direction of the eigenvector. 

And then based on the ranked eigenvalues:

Linearity | Lλ |  (λ_1- λ_2)/λ_1 
Planarity | Pλ |  (λ_2- λ_3)/λ_1 
Scatter | Sλ |  ( λ_3)/λ_1 
Omnivariance | Oλ |  ∛(λ_1 λ_2 λ_3 )
Eigentropy | Eλ |  -λ_1  ln⁡(λ_1 )-λ_2  ln⁡(λ_2 )-λ_3  ln⁡(λ_3 )
Sum of eigenvalues | ∑λ |  λ_1+λ_2+λ_3
Curvature | Cλ |  λ_3/(λ_1+λ_2+λ_3 )

These metrics are intensively used for point cloud classification in the urban environment ([Martin Weinmann work](https://www.ipf.kit.edu/english/staff_weinmann_martin.php) publications), but also in TLS data processing to extract e.g. branch structure of the trees [Bremer et al, 2013](https://www.sciencedirect.com/science/article/pii/S0924271613000774). I think that these features would be beneficial to calculate at the voxel level, but calculating in a grid_metrics() as an area-based approach, planarity helps to differentiate buildings from non-buildings.

Jean-Romain, 2019-02-02:
Thank for the explanations and the references.

I can't see the pertinence of such metrics in an area-based approach but I'm not the judge of what people should do. In you specific example of building I think the future function `lascoplanar` (#209) is more appropriated (but it basically rely on similar computations). Anyway such function can be apply at the plot level (`lasmetrics`), pixel level (`grid_metrics`), voxel level (`grid_metrics3d)` and tree level (`tree_metrics`). As long as it works users can try whatever they want.

What I can do is simply a cpp implementation of `eigen_values`:

```r
Rcpp::sourceCpp(code = "
#include <RcppArmadillo.h>
// [[Rcpp::depends(RcppArmadillo)]]

// [[Rcpp::export]]
SEXP eigen_value(arma::mat A) {  
arma::mat coeff;
arma::mat score;
arma::vec latent;
arma::princomp(coeff, score, latent, A);
return(Rcpp::wrap(latent));
}")

eigen_metrics = function(X,Y,Z)
{
  xyz = rbind(X,Y,Z) 
  eigen_m = eigen_values(xyz)
  ...
}
```

This gives a 4-fold speed-up to your function. If you are comfortable with package development you can fork the package and add these new functions + documentation then make a pull request. Otherwise I will do it myself but later because I must read your references first.

Jean-Romain, 2019-02-17:
I'm almost done. I need to write the documentation. Did you noticed you made an error in your code? `rbind` should be `cbind`.

komazsofi, 2019-02-17:
Oh, you are right! Thanx!!! I am just re-running my calculations ;-)

Jean-Romain, 2019-02-17:
Done and avaible in lidR v2.1.0 in `devel` branch.

```r
file <- system.file("extdata", "Megaplot.laz", package="lidR")
las  <- readLAS(LASfile)
col  <- height.colors(50)
eigm <- grid_metrics(las, .stdshapemetrics, 3)
plot(eigm, col = col)
```

[image]

closed 2019-02-17 by Jean-Romain

bi0m3trics, 2019-02-18:
Looks like stdshapemetrics is missing the `#' @export` in `utils_metrics.r` - unless that's how you intended it to be used.

Jean-Romain, 2019-02-18:
Exact it is missing. But `.stdshapemetrics` works.

Jean-Romain, 2019-02-19:
@bi0m3trics  reported an additional issue by email. @komazsofi if you already installed, please reinstalll

commit 9b70d0b5 (2019-02-17): Add stdshapemetrics (close #217)