r-lidar/lidR #219: Invalid data: ScanAngleRank greater than 90

pobsteta, 2019-02-10:
Hi Jean-Romain,

I don't know if you speak french, so i describe may problem with my english. I have 9 las files, if i take each file individualy, i can normalize it but with a catalog file i obtain this error :"Invalid data: ScanAngleRank greater than 90" why ?
@+

Jean-Romain, 2019-02-11:
Please give me a reproducible example. I can't help you if you only say "it does not work as expected". 
Is it an error or a warning? Is is when reading the file or after normalization? I don't understand why you have such message since this message has been added in the next release of the package. Which version of the package are you using?

pobsteta, 2019-02-11:
j'ai lancé ce script :

pobsteta, 2019-02-11:
ctg = catalog(rep, recursive=TRUE)
# normalize las catalog
opt_output_files(ctg) <- paste0(rep, "/outputs/{ID}_norm")
opt_chunk_size(ctg) <- 500
opt_chunk_buffer(ctg) <- 30
opt_cores(ctg) <- 7
# Streaming way
new_ctg <- lasnormalize(ctg, tin())

Jean-Romain, 2019-02-11:
Is it an error or a warning? Is is when reading the file or after normalization? Which version of the package are you using?

Jean-Romain, 2019-02-11:
Can you reproduce with:

```r
ctg = catalog(rep, recursive=TRUE)
las = readLAS("ctg$filename[1]")
```

Is is an error or a warning? If it is a warning there is no bug. Pay attention to it and if it annoy you use `suppressWarnings`. If it is a error I made a mistake in the code it should be a warning.

Jean-Romain, 2019-02-11:
Ho I got it. When reading the file you have nothing wrong because lidR 2.0.1 does not take advantage of `rlas 1.3.1` yet. But when writing the file you have an error because `writeLAS` call `rlas::write.las` that make stronger assertions.

Minimal reproducible example:

```r
library(lidR)
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
las = readLAS(LASfile)
las@data$ScanAngleRank[1] <- 91L
f = tempfile(fileext = ".las")
writeLAS(las, f)
```

Sadly I can't help you. I do agree that it could technically work without error for angles up to 128 degrees. I will fix that in `rlas 1.3.2` but for now it just doesn't work because the `rlas` package strongly respect the las specifications that state that the `ScanAngleRank` range between -90 and 90 degrees.

The problem with dataset that don't respect the specification is that it generates unexpected bug. You should first ask to your provider why you received incorrect data.

Jean-Romain, 2019-02-11:
See https://github.com/Jean-Romain/rlas/issues/33. I will write a patch tomorrow

closed 2019-02-11 by Jean-Romain

pobsteta, 2019-02-12:
Hi, thanks for explaination. I have the same error with
> ctg = catalog(rep, recursive=TRUE)
Warning message:
In fun(libname, pkgname) :
  rgeos: versions of GEOS runtime 3.7.0-CAPI-1.11.0
and GEOS at installation 3.6.2-CAPI-1.10.2differ
> las = readLAS(ctg$filename[1])
> plot(ctg, mapview = TRUE)
> # normalize las catalog
> opt_output_files(ctg) <- paste0(rep, "/outputs/{ID}_norm")
> opt_chunk_size(ctg) <- 500
> opt_chunk_buffer(ctg) <- 30
> opt_cores(ctg) <- 7
> # Streaming way
> new_ctg <- lasnormalize(ctg, tin())
865 points below 0 found.
Erreur : Invalid data: ScanAngleRank greater than 90

Jean-Romain, 2019-02-12:
This is the same code than the previous one. Why do you expect another behavior? The point is that you can read your files but you can't write them because they are invalid. So your problem will happen in almost all `las*` function when processing a catalog.

By the way there is not "streaming" here. All the data are loaded in memory

pobsteta, 2019-02-12:
Hi i load rlas 1.3.1 and lidR 2.0.1
with this filter 
opt_filter(ctg) <- "-drop-scan-angle '>=91'"
i obtain the same error
and with this 
opt_output_files(ctg) <- paste0(rep, "/outputs/{ORIGINALFILENAME}_normalized")
Error in eval(parse(text = text, keep.source = FALSE), envir) : 
  objet 'ORIGINALFILENAME' introuvable
And if i filter las files with las2las and option --drop-scan-filter it work ?

Jean-Romain, 2019-02-12:
I said you to install `rlas 1.3.2` from github to fix your problem. `rlas 1.3.1` is the current version on CRAN.

The idea could work but `-drop-scan-angle` does not exist. See possible options with `rlas:::lasfilterusage`

Again I can't help you with no reproducible code. I can't guess what you wrote before and after that. Well, actually I can make a guess. You probably wrote something like:

```r
opt_chunk_size(ctg) <- 500
opt_output_files(ctg) <- paste0(rep, "/outputs/{ORIGINALFILENAME}_normalized")
```

And indeed it failed. Please refer to the documentation to understand why.

If you have question about `lidR` please ask on https://gis.stackexchange.com/ with the tag `lidr`. Github is a place to report bugs. On https://gis.stackexchange.com/ your questions will be indexed by search engine and useful for everybody.