r-lidar/lidR #223: lasmergespatial bug?

NatKatUP, 2019-02-18:
Hi,
I have a bug where lasmergespatial(las, polygon, "ID") assigns the SpatialPolygonDataframe's attribute "ID" not to the points that lay within the polygons, but to the first length(polygon) of points in subsequent order.
They are in the same projection and lasclassify worked for the same kind of las dataset and polygon structure.
I would upload the las-file, but it's kinda big, so maybe someone comes up with a solution without. 

Thanks,
NatKat

Jean-Romain, 2019-02-18:
If you can't share the file, at least show me the code and an image of the output because I don't understand what is going wrong

Jean-Romain, 2019-02-18:
Easy to reproduce. Bug confirmed. I should be able to fix it within few minutes.

```r
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
shp     <- system.file("extdata", "lake_polygons_UTM17.shp", package = "lidR")

las     <- readLAS(LASfile, select = "xyz")
lakes   <- shapefile(shp)

las <- lasmergespatial(las, lakes, "ANY")
table(las@data$ANY)

las <- lasmergespatial(las, lakes, "LAKENAME_1")
table(las@data$LAKENAME_1)
```

Jean-Romain, 2019-02-18:
Fixed

closed 2019-02-18 by Jean-Romain

NatKatUP, 2019-02-20:
I was out of office for two days, thanks for solving this problem so fast! 
And thanks again for the comprahensive package, I am writing my master's thesis and use lidR extensively.

NatKatUP, 2019-02-20:
Unfortunately there is another problem with lasmergespatial: 
I have a polygon containing the ground projected crown shapes of a tree stand and want to give my point cloud points that belong to the trees their IDs. 
It seems like my polygons change their shape within the funtion. It looks like the polygons would maybe get rasterized and resampled, at least their shape is much simpler and if i color the point cloud by ID in the RGL viewer, you see that not all points got the id (see screenshot).
[image]
I attach a las subset and my shape file.
[shapes.zip](https://github.com/Jean-Romain/lidR/files/2884180/shapes.zip)
[test_subset.zip](https://github.com/Jean-Romain/lidR/files/2884181/test_subset.zip)

Jean-Romain, 2019-02-20:
Could you please open a new issue with the data, the image, the explanation + a reproducible code. Thank you.

commit d7fdb16f (2019-02-18): Fix #223