r-lidar/lidR #224: Scale Bar when displaying point cloud.

Selenachav, 2019-02-18:
Is there a way to get a scale bar that shows the color and its corresponding height when I process and display my point cloud? or is this more of an R Studio question to ask?

Jean-Romain, 2019-02-18:
Well this is actually an `rgl` question. The answer is *probably yes but I don't know how*.

One simple solution is:

```r
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
las     <- readLAS(LASfile, select = "xyz")

plot(las)
axis3d("z", pos = c(0, 0, NA), col = "white" )
```

But to add a scale color bar it is more complex. Here an ugly solution:

```r
plot(las)
rgl::bgplot3d( 
  fields::image.plot(
    legend.only = TRUE, 
    legend.args = list(text='Z'), 
    zlim = c(min(las$Z), max(las$Z)), 
    col = height.colors(50)
  )
) 
```
If you find a way (link to forum, source code or any) to add this feature I will add it for sure.

Selenachav, 2019-02-18:
Thank you.
I have posted my question on rgl forum as well.
But your "ugly solution" worked out well for what I need the scale bar for.

Also, I have not really used the lidR package since it been updated when you refer to "system.file" is that just the location of where my las file is?

Jean-Romain, 2019-02-18:
`system.file` returns user specific path to  dataset provided with the package. To me its `"/home/jr/R/x86_64-pc-linux-gnu-library/3.5/lidR/extdata/Megaplot.laz"` but it is something else for you.

Jean-Romain, 2019-02-21:
Here a *not too bad* solution I could implement.

```r
library(lidR)
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
las     <- readLAS(LASfile, select = "xyzi")

trim = 90
grad = heat.colors(50)
col  = "Intensity"

f = tempfile(fileext = ".png")
png(f, 1920, 1080, bg = "black")
par(col.axis = "white")
fields::image.plot(
  legend.only = TRUE,
  legend.width = 2.5,
  legend.args = list(text='Z', col = "white"), 
  zlim = c(min(las@data[[col]]), min(max(las@data[[col]]), trim)), 
  col = grad)
dev.off()

plot(las, color = "Intensity", trim = trim, colorPalette = grad)
rgl::bg3d(texture = f, col = "white")
```

Jean-Romain, 2019-02-21:
I added an argument `legend = TRUE`

```r
plot(las, axis = TRUE, legend = TRUE)
```

Work best in full screen with a 1920 x 1080 monitor

closed 2019-02-21 by Jean-Romain