r-lidar/lidR #227: [2.0.1] Errors in processing catalog

spono, 2019-02-20:
Ciao JR,
I noticed some strange behaviours using `grid_metrics` on a `catalog `(both as separate tiles and big single file):
-	Calculating the LAD I get the error
```
Error: nrow(coords) > 0 is not TRUE
In addition: Warning messages:
1: In points2grid(points, tolerance, round) :
  grid has empty column/rows in dimension 1
2: In points2grid(points, tolerance, round) :
  grid has empty column/rows in dimension 1
```

-	Calculating the VCI I get 
`Error: in .local(x, y, ...) : extents do not overlap`

They stop at different percentage of progress (7% and 69%, respectively) so it’s not a matter of the same area. Unfortunately I don’t any other idea where to look for the error. Any suggestion?

Many thanks in advance!

Jean-Romain, 2019-02-20:
At least give me a the code that generated this error please.

spono, 2019-02-20:
yep, sorry. Here it is:

```
ctg = catalog(LASfolder)
opt_select(ctg) = 'xyz'
opt_filter(ctg) = '-drop_z_below 0 -drop_z_above 45 -keep_first'
opt_chunk_size(ctg) <- 1000
opt_wall_to_wall(ctg) <- TRUE
opt_cores(ctg) <- 6L#cpus
opt_output_files(ctg) <- file.path(tempFolder,"clip")
opt_laz_compression(ctg) <- TRUE

rast.resol = 100

r = grid_metrics(ctg, LAD(Z, dz = 5), res=rast.resol)
crs(r) <- new_projectCRS
writeRaster(r, filename=file.path(outFolder,'raster', paste0('LAD_',rast.resol,'m.tif')),
            format="GTiff", options="COMPRESS=LZW", overwrite=TRUE)

r = grid_metrics(ctg, VCI(Z, zmax=45, by = 2), res=rast.resol)
crs(r) <- new_projectCRS
writeRaster(r, filename=file.path(outFolder,'raster', paste0('VCI_',rast.resol,'m.tif')),
            format="GTiff", options="COMPRESS=LZW", overwrite=TRUE)

```

Jean-Romain, 2019-02-20:
I don't think it is an error with `LAD` or `VCI`. I think it is an error with the `raster`. You probably found a limit case with some chunks that have 1 cells or any weird case like that. Please show me the chunk pattern with `plot(ctg, chunk = TRUE)`. I think it is a bug that will be hard to find without the exact data.

spono, 2019-02-20:
[image]

Jean-Romain, 2019-02-20:
Is it a 18 x 22 km tile? :scream: Is it possible to share? Which chunk failed? Does it failed with a different tiling size? e.g 1500?

spono, 2019-02-20:
> Is it a 18 x 22 km tile?

ehm...yes. I tried this ugly solution to check if the error was possibly related to something in the tiling scheme or not

> Is it possible to share?

quite hard (mostly for data privacy)

> Which chunk failed?

different one for `LAD `and `VCI`

> Does it failed with a different tiling size?

I didn't try a different tile size because it works with a different pixel resolution (...weird...)

Jean-Romain, 2019-02-20:
> I didn't try a different tile size because it works with a different pixel resolution (...weird...)

Yes this is what I guessed. You are probably in a kind of limit case that is almost impossible to predict to me. I need people encountering bugs to fix them because I have never found a way to generate weird limits cases myself.

Luckily I ran your `VCI` example on one of my catalog and I reproduced. Let investigate.

Jean-Romain, 2019-02-20:
I have found one of the two errors. The error with `VCI` occurred for a chunk that was only the buffer (no actual data in the chunk but the chunk was computed anyway because the buffer lied on a tile). When removing the buffer nothing was retained.

I can fix it by better filtering the valid chunks. For the other issue I'm still investigating

Note that my internal version of lidR have a better error handling. This is easier to find errors :wink:

```
Processing [=============>--------------------------]  36% (144/398) eta:  3m
An error occurred when processing the chunk 144. Try to load this chunk only:
chunk <- readRDS(/tmp/RtmpfdPzX7/catalog_apply_chunk_144.rds)
las <- readLAS(chunk)

 Error in .local(x, y, ...) : extents do not overlap 
```

spono, 2019-02-20:
> Note that my internal version of lidR have a better error handling. This is easier to find errors

I noted...now I simply need to learn how to use it properly in order not to bother you ;D
(lesson from today:
```
chunk <- readRDS(/tmp/RtmpfdPzX7/catalog_apply_chunk_144.rds)
las <- readLAS(chunk)
```
)
thanks!

Jean-Romain, 2019-02-20:
You don't bother me. You found a real bug

Jean-Romain, 2019-02-20:
I have fixed the issue when processing my own catalog. But your issue, even if it is the same error, does not occurs for the same reason. According to your plot, you don't have any chunk that overlap your tiles only with the buffer. All your chunks actually belong on your tile.

It means that your dataset is not rectangular and you found in a very limit case where the chunks overlap the data only with the buffer but we can't guess it before to actually read the data. I wrote a fix for this case but it is not easy to test it myself. Please confirm that it fixed your `VCI` case.

Jean-Romain, 2019-02-20:
For `LAD` this is not a bug. You just can't use `LAD` in  `grid_metrics`. It is not a metric.

Jean-Romain, 2019-02-20:
I added an error for such cases

```r
r = grid_metrics(las, LAD(Z, dz = 5), res=rast.resol)
 Erreur : Duplicated pixel founds. At least one of the metrics was not a number. Each metric should be a single number. 
```

closed 2019-02-20 by Jean-Romain