r-lidar/lidR #228: lasmergespatial changes polygon shape

NatKatUP, 2019-02-20:
Unfortunately there is another problem with lasmergespatial:
I have a polygon containing the ground projected crown shapes of a tree stand and want to give my point cloud points that belong to the trees their IDs.
It seems like my polygons change their shape within the funtion. It looks like the polygons would maybe get rasterized and resampled, at least their shape is much simpler and if i color the point cloud by ID in the RGL viewer, you see that not all points got the id (see screenshot).
[image]
 I attach a las subset and my shape file.
[test.zip](https://github.com/Jean-Romain/lidR/files/2884686/test.zip)

`library(rgdal)
testlas <- readLAS("test_subset.las")
testPoly <- readOGR(dsn = ".", layer = "EN18012_crownsPoly") # attribute testPoly@data$ID is the trees ID

plot(SpatialPoints(testlas@data[,1:2]), pch = 16, cex = 0.2)
plot(testPoly, add= TRUE, border = "red")

testlas <- lasmergespatial(testlas, testPoly, "ID") # assign ID's

plot(testlas, color = "ID") # look at the top view and compare with plot
`

Jean-Romain, 2019-02-20:
To me it looks like a problem with floating point precision. You coordinates are very high number thus the issue is amplified. Moreover your polygons are tiny.  The error is almost invisible for wide polygons but is very visible here. Let make a test.

```r
library(lidR)

shp  <- shapefile("EN18012_crownsPoly.shp")
las  <- readLAS("test_subset.las")

# Shift
xmin <- min(las$X)
ymin <- min(las$Y)

# Shift everything to (0,0)
las$X <- las$X - xmin
las$Y <- las$Y - ymin
shp   <- shift(shp, x = -xmin, y = -ymin)

las <- lasmergespatial(las, shp, "ID")
plot(las, color = "ID")
```

It looks good. That is very annoying because the bug should occurs in `rlas` as well since both `lidR` and `rlas` rely on `boost::geometry::covered_by`. 

At least you can use the following workaround waiting for a real fix.

```r
lasmergespatialspdf = function(las, spdf, attr = NULL)
{
  xmin  <- min(las$X)
  ymin  <- min(las$Y)
  las$X <- las$X - xmin
  las$Y <- las$Y - ymin
  spdf  <- shift(spdf, x = -xmin, y = -ymin)
  las   <- lasmergespatial(las, spdf, attr)
  las$X <- las$X + xmin
  las$Y <- las$Y + ymin
  return(las)
}
```

Jean-Romain, 2019-02-20:
I found the origin of the trouble. When parsed as WKT string a to small number of digit was used. I will fixed that within few minutes

```r
spshp    <- shapefile("EN18012_crownsPoly.shp")
sfshp     <- sf::st_as_sf(spshp)
sptree1  <- spshp[1,]
sftree1   <- sfshp[1,]

# Reproduce lidR behavior
wkt <- sf::st_as_text(sftree1$geometry)
wktree1 <- rgeos::readWKT(wkt)

plot(sptree1)
plot(sftree1["ID"])
plot(wktree1)

# Use more digits
wkt <- sf::st_as_text(sftree1$geometry,  digits = 14)
wktree1 <- rgeos::readWKT(wkt)

plot(sptree1)
plot(sftree1["ID"])
plot(wktree1)
```

NatKatUP, 2019-02-20:
Thanks, this works for me.

Jean-Romain, 2019-02-20:
Fixed in v2.0.2

closed 2019-02-20 by Jean-Romain

NatKatUP, 2019-02-20:
Also lasclip produces errors: "No point found for within POLYGON".
Probably for the same reason.
It works with
`
lasclipspdf = function(las, spdf){
  xmin  <- min(las$X)
  ymin  <- min(las$Y)
  las$X <- las$X - xmin
  las$Y <- las$Y - ymin
  spdf  <- shift(spdf, x = -xmin, y = -ymin)
  las   <- lasclip(las, spdf)
  las$X <- las$X + xmin
  las$Y <- las$Y + ymin
  return(las)
}
`

Jean-Romain, 2019-02-20:
I also fixed `lasclip` in the same commit. If you still encounter troubles it is another issue.

commit 91f5c8d1 (2019-02-20): Fix #228