r-lidar/lidR #231: grid_metrics issue

muzo583, 2019-03-02:
Hi all,
I have tried grid_metrics help page code. But it generated error. do you have any idea?

```
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
las = readLAS(LASfile)
colors = height.colors(50)

# Canopy surface model with 4 m^2 cells
metrics = grid_metrics(las, max(Z), 2)
plot(metrics, col = colors)

# Mean height with 400-m^2 cells
metrics = grid_metrics(las, mean(Z), 20)
plot(metrics, col = colors)

# Define your own new metrics
myMetrics = function(z, i)
{
  metrics = list(
     zwimean = sum(z*i)/sum(i), # Mean elevation weighted by intensities
     zimean  = mean(z*i),       # Mean products of z by intensity
     zsqmean = sqrt(mean(z^2))  # Quadratic mean
   )

   return(metrics)
}

metrics = grid_metrics(las, myMetrics(Z, Intensity))

plot(metrics, col = colors)
plot(metrics, "zwimean", col = colors)
plot(metrics, "zimean", col = colors)
plot(metrics, "zsqmean", col = colors)
```

ERRO is here,
```
metrics = grid_metrics(las, myMetrics(Z, Intensity))
Error in lazyeval::f_capture(func) : Promise has already been forced
```

I thought, may it occurred  due to the library error, I have installed lazyeval library but no luck.

platform       x86_64-pc-linux-gnu                                     
version.string R version 3.5.2 (2018-12-20)
Rstudio Version 1.1.463

Jean-Romain, 2019-03-03:
I don't have such issue running your code. Please retry your code in a fresh session and confirm it is reproducible.

Anyway the following syntax should be preferred everywhere in lidR.

```r
metrics = grid_metrics(las, ~myMetrics(Z, Intensity))
```

muzo583, 2019-03-03:
Hmm. It is interesting. 
As you pointed, when I clearly run the code it works. But when I run this,
```
myMetrics = function(z)
{
  mm =list(out = boxplot(z,plot = FALSE, outline = TRUE))
  #ind = which(!veri2$Z %in% out, arr.in= TRUE)

  return(mm)
}
metrics = grid_metrics(veri, myMetrics(Z), 100)
```
this code gives an error. After, I run the help page code it produces the same error again.

Thanks Romain.

Jean-Romain, 2019-03-03:
I can't reproduce. 

But as I said use `~myMetrics(Z)`. In the next release (v2.1) I changed the documentation to add the `~` in the examples. Former syntax will still work but including the ~ everywhere will avoid many problems of this kind.

Btw your code gives an error in my version (you don't have the same error yet and your error is more cryptic (see also #227)).

```r
metrics = grid_metrics(las, myMetrics(Z), 100)
#> Erreur : Duplicated pixels found. At least one of the metrics was not a number. Each metric should be a single number. 
```

What you are trying to compute is incorrect. If you don't understand why please ask on https://gis.stackexchange.com/

spono, 2019-06-21:
Ciao JR,
sorry for re-opening but, looking for the abovementioned `lazy_eval `error, this thread was among teh few useful things that popped up.

I'm getting the same issue rasterizing my own function.

`r4 = grid_metrics(las, ~vofun(X,Y,Z), res=1)`

Does this mean that it is not able to evaluate the `~vofun(X,Y,Z)` argument or that within the output list there might be an issue? in this last case, the presence of `NA` are an issue for the rasterization process?

spono, 2019-06-22:
Ciao JR,
sorry for re-opening but, looking for the abovementioned `lazy_eval `error, this thread was among the few useful things that popped up.

I'm getting the same issue rasterizing my own pretty long (138 vars) function.

`r = grid_metrics(las, ~vofun(X,Y,Z), res=1)`

Does the error mean that the algorithm is not able to evaluate the `~vofun(X,Y,Z)` argument or that within the function there might be any other kind of issue? in this last case, the presence of `NA ` are an issue for the rasterization process?

I don't see any other potential issue and still don't find useful info online on how to "debug" it.
thanks in advance

right now I found a workaround applying the same function a little tweaked to work on `catalog` and it works fine...but obviously then I have to rasterise every single variable

Jean-Romain, 2019-06-22:
Please provide a reproducible example.

Jean-Romain, 2019-06-22:
Reproducible example sent by email. Error confirmed.

Jean-Romain, 2019-06-25:
I found your issue. As mentioned in the documentation you should always use the `~` in `grid_metrics` as well as in similar function. Without the `~` it works only at the first level (global_env). Do no ask me why. Here is your issue.

```r
metr = lasmetrics(impLAS, ~stdmetrics_z(Z, dz=0.5))
```

closed 2019-06-25 by Jean-Romain