r-lidar/lidR #234: Can't save LAS file after select and filter on readLAS

jakubchr, 2019-03-15:
lidR 2.0.2

LAS file summary: 
```
class        : LAS (LASF v1.4)
point format : 7
memory       : 107.1 Mb 
extent       :7497590, 7498410, 5788790, 5789610 (xmin, xmax, ymin, ymax)
coord. ref.  : NA 
area         : 670422.1 units² (convex hull)
points       : 4011573 points
density      : 5.98 points/units²
names        : X Y Z Intensity 
File signature:           LASF 
File source ID:           0 
Project ID - GUID:        00000000-0000-0000-0000-000000000000 
Version:                  1.4
System identifier:        LAStools (c) by rapidlasso GmbH 
Generating software:      lasheight (181108) commercial 
File creation d/y:        12/2019
header size:              375 
Offset to point data:     815 
Num. var. length record:  1 
Point data format:        7 
Point data record length: 40 
Num. of point records:    4011573 
Num. of points by return: 
Scale factor X Y Z:       0.01 0.01 0.01 
Offset X Y Z:             0 0 0 
min X Y Z:                7497590 5788790 -1.58 
max X Y Z:                7498410 5789610 38.92 
Variable length records: 
   Variable length record 1 of 1 
       Description:  RIEGL Extra Bytes 
       Extra Bytes Description:
          Amplitude: Echo signal amplitude [dB]
          Pulse width: Full width at half maximum [ns]
```
Here's the code that I'm trying to run:
```
las <- readLAS("7497600_5788800.las",select = "xyzi", filter = "-keep_class 4 5")
writeLAS(las, "filtered_selected_las.las")
```
I receive error:

> Error: Invalid file: the header describes extra bytes attributes that are not in the data.

I would like to save such filtered and selected file. I don't know whether it is prohibited  deliberately.
Cheers! :)

Jean-Romain, 2019-03-15:
Well, in this particular example the message could be a warning. But you can find a workaround by removing the extrabytes from the header

```r
las@header@VLR <- list()
```

Jean-Romain, 2019-03-15:
Not to myself: 

- [ ] if `select` explicitly removes the extrabytes, `readLAS` should also remove the extrabytes.

jakubchr, 2019-03-15:
Ok, it allows me to save but it seems like writeLAS ignores selection and filtering.
It saves file with all attributes (beside xyzi)

```
writeLAS(las, "filtered_selected_las.las")
las_new = readLAS("filtered_selected_las.las")
```
and the summary for las new:

```
class        : LAS (LASF v1.4)
point format : 7
memory       : 397.9 Mb 
extent       :7497590, 7498410, 5788790, 5789610 (xmin, xmax, ymin, ymax)
coord. ref.  : NA 
area         : 670422.1 units² (convex hull)
points       : 4011573 points
density      : 5.98 points/units²
names        : X Y Z gpstime Intensity ReturnNumber NumberOfReturns ScanDirectionFlag EdgeOfFlightline Classification ScannerChannel Synthetic_flag Keypoint_flag Withheld_flag Overlap_flag ScanAngle UserData PointSourceID R G B 
File signature:           LASF 
File source ID:           0 
Project ID - GUID:        00000000-0000-0000-0000-000000000000 
Version:                  1.4
System identifier:         
Generating software:      rlas R package 
File creation d/y:        0/2019
header size:              375 
Offset to point data:     375 
Num. var. length record:  0 
Point data format:        7 
Point data record length: 36 
Num. of point records:    4011573 
Num. of points by return: 
Scale factor X Y Z:       0.01 0.01 0.01 
Offset X Y Z:             0 0 0 
min X Y Z:                7497590 5788790 -1.58 
max X Y Z:                7498410 5789610 38.92 
Variable length records:  void
```

Jean-Romain, 2019-03-15:
A las file contains all the attributes. What you can change is the point format record. See the [las specifications](https://www.asprs.org/a/society/committees/standards/LAS_1_4_r13.pdf). For a more comprehensive answer please ask on https://gis.stackexchange.com/.

Jean-Romain, 2019-03-20:
Enhanced in lidR v2.1

``` r
library(lidR)
f <- system.file("extdata", "extra_byte.laz", package = "rlas")
las <- readLAS(f, select = "xyzi")
#> 'Amplitude' extrabytes attribute has been removed from the header because it has not been loaded.
#> 'Pulse width' extrabytes attribute has been removed from the header because it has not been loaded.
las <- readLAS(f, select = "xyzi1")
#> 'Pulse width' extrabytes attribute has been removed from the header because it has not been loaded.
```

<sup>Created on 2019-03-20 by the [reprex package](https://reprex.tidyverse.org) (v0.2.1)</sup>

commit 6fe87cc8 (2019-03-20): Fix #234
closed 2019-03-20 by Jean-Romain