r-lidar/lidR #238: Error in grid_canopy

Eduardo-Tusa, 2019-03-26:
Hi,
When I use a for loop:
# *****************************************************************************************
myres = c(1.0,0.95,0.9,0.85,0.8,0.75,0.7,0.65,0.6,0.55,0.5,0.45,0.4,0.35,0.3,0.25)
for (ires in myres)
{
  chm = grid_canopy(las, res = ires, p2r())
  plot(chm)
}
# *****************************************************************************************
I got an error:
# *****************************************************************************************
Error in algorithm(las, layout) : 
  Internal error: matrix returned at the C++ level does not match with the layout. Please report this bug.
# *****************************************************************************************
But, when I write like this:
# *****************************************************************************************
chm = grid_canopy(las, res = 0.2, p2r())
# *****************************************************************************************
I have no problem at all...
--
 
> | >
>

Jean-Romain, 2019-03-29:
This error should not happen but it still happens in very rare cases with some specific dataset and resolution. Please find the resolution that failed and share the point cloud or a subset of the point cloud that enable to reproduce. Thanks.

Jean-Romain, 2019-04-12:
No news ?

alekszaf, 2019-08-05:
I have encountered the same problem whenever I pick resolution lower than 0.4-0.5, depending on the plot. I have attached two of the plots I am processing, plot C shows the error if res < 0.4, plot E res <0.5.

[hlw_june_plots.zip](https://github.com/Jean-Romain/lidR/files/3468526/hlw_june_plots.zip)

.

Jean-Romain, 2019-08-05:
Thank you, I identified the issue. I'm working on a fix.

Jean-Romain, 2019-08-05:
Fixed.

commit 15229363 (2019-08-05): Fix #238
closed 2019-08-05 by Jean-Romain