r-lidar/lidR #241: Best way to extract density from summary object?

bw4sz, 2019-04-08:
Small request. Can extracting the density from a lidR object be made easier?

```
> r
class        : LAS (LASF v1.3)
point format : 3
memory       : 3.8 Mb 
extent       :252874, 252914, 4104694, 4104734 (xmin, xmax, ymin, ymax)
coord. ref.  : +init=epsg:32611 +proj=utm +zone=11 +datum=WGS84 +units=m +no_defs +ellps=WGS84 +towgs84=0,0,0 
area         : 1598.113 m² (convex hull)
points       : 42588 points
density      : 26.65 points/m²
names        : X Y Z gpstime Intensity ReturnNumber NumberOfReturns ScanDirectionFlag EdgeOfFlightline Classification Synthetic_flag Keypoint_flag Withheld_flag ScanAngle UserData PointSourceID R G B
```
I can see it, but not obvious in the summary or @header slots.

```
> r@header
File signature:           LASF 
File source ID:           0 
Project ID - GUID:        00000000-0000-0000-0000-000000000000 
Version:                  1.3
System identifier:        LAStools (c) by rapidlasso GmbH 
Generating software:      lascolor (160730) commercial 
File creation d/y:        104/2018
header size:              235 
Offset to point data:     1100 
Num. var. length record:  2 
Point data format:        3 
Point data record length: 34 
Num. of point records:    42588 
Num. of points by return: 37000 4128 1232 212 16 
Scale factor X Y Z:       0.001 0.001 0.001 
Offset X Y Z:             250000 4100000 0 
min X Y Z:                252874 4104694 -0.76 
max X Y Z:                252914 4104734 10.637 
Variable length records: 
   Variable length record 1 of 2 
       Description:  by LAStools of rapidlasso GmbH 
       Tags:
          Key 1024 tiff_tag_location 0 count 1 value offset 1 
          Key 3072 tiff_tag_location 0 count 1 value offset 32611 
          Key 3076 tiff_tag_location 0 count 1 value offset 9001 
          Key 4099 tiff_tag_location 0 count 1 value offset 9001 
   Variable length record 2 of 2 
       Description:  OGC COORDINATE SYSTEM WKT 
```
Can the summary() return what is printed in case the user wants to extract that value?

Jean-Romain, 2019-04-08:
Exact. Just pay attention to the fact that `area` returns the convex hull area and thus might underestimate the area in non convex point clouds. I do agree that I could export some simple tools like that. If you have other suggestion please add them I'll try to do it sooner or later

Jean-Romain, 2019-04-12:
Added `density()` and `npoints()` in v2.1.0

``` r
library(lidR)
file   <- system.file("extdata", "Megaplot.laz", package="lidR")
las    <- readLAS(file, filter = "-keep_first")
header <- readLASheader(file)
ctg    <- catalog(file)

npoints(las)
#> [1] 55756
npoints(header)
#> [1] 81590
npoints(ctg)
#> [1] 81590

density(las)
#> [1] 1.049783
density(header)
#> [1] 1.535576
density(ctg)
#> [1] 1.535576
```

<sup>Created on 2019-04-12 by the [reprex package](https://reprex.tidyverse.org) (v0.2.1)</sup>

closed 2019-04-12 by Jean-Romain