r-lidar/lidR #244: extracting cloth from cloth simulator filter

mikoontz, 2019-04-11:
First, thanks so much for this package. I'm using it as the primary way to extract meaningful data from drone-derived, structure-from-motion point clouds. As an aside, I'm finding your `lmfx` algorithm to be the best at detecting trees in my study sites, and about 100 times faster than the next-best-algorithm (`li2012`). As part of the paper I'm writing, I'm testing several (https://github.com/mikoontz/local-structure-wpb-severity).

Here's my feature request: Would it be possible to extract the "cloth" that is used to classify the point cloud into ground/non-ground as a `raster` object? 

My [current workflow](https://github.com/mikoontz/local-structure-wpb-severity/blob/master/data/data_carpentry/create-canopy-height-models.R) is to use the `lidR::lasground()` function with the `algorithm =` argument as `csf()` (with a few tuned parameter values) to classify ground/non-ground, then to use `lidR::grid_terrain()` with the `algorithm = ` argument set to `tin()` to create a digital terrain model. During the interpolation process, there are sometimes some DTM artifacts associated with the uneven coverage of classified "ground" points within a scene, and the cloth itself might be a better representation of the terrain underneath the vegetation compared to the spatial interpolation between the "ground" points. I was finding the smoother cloth to be a better representation of the DTM when using the cloth simulator filter algorithm as implemented in CloudCompare (and then exporting the cloth itself as a raster layer).

I'm now also seeing the `keep_lowest = ` argument, which might start to smooth out some of these artifacts so I can play around with that too.

Thanks for considering it!

Jean-Romain, 2019-04-11:
Dear user. You should better ask your question to [RCSF](https://github.com/Jean-Romain/RCSF) and/or in the original library [CSF](https://github.com/jianboqi/CSF). lidR relies on this package to make a point-cloud based segmentation of ground point and will not provided other tools. But depending on what the CFS library has internal it might or might not be possible to add such feature in RCSF (I don't now myself actually)

closed 2019-04-11 by Jean-Romain

mikoontz, 2019-04-11:
Oh, wow, I didn't realize that it was an external dependency. I'll check with them; thanks!