r-lidar/lidR #246: Create las object from xyz coordinates

nFrechen, 2019-04-18:
Great package! I was wondering what would be the best way to create a las object from plain xyz data and then save it to a las file. I guess the referencing, projection, bounding box calculation etc. has to be done. I am shure it is already possible but I have no clear idea about how to go about it and find no documentation or mention that this is a thing you can do.

Background is that I have a simple (and cheap) lidar scanner at work that basicly gives me distances for certain angles:
https://www.robotshop.com/uk/small-ls02-solid-state-2d-lidar-86-4m.html
I move it over a plot with a crane and record the travel distance. Hence I can calcullate xyz coordinates. But how to cast these data in a las file now?

I am wondering if you might be interested to write a tutorial about how to do that with the lidR package. But I don't know how generally intresting this might be. Seems to me that most lidar scanners already output las files. But maybe someone wants to work with data from an unusual source like me...

spono, 2019-04-18:
Hi,
if you have a table with xyz info, you can simply create a las with:

```
xyz = read.table(...)
las = LAS(data=xyz)
```

and then you export. The other fields will be automatically created (empty) to attain at the ASPRS specs.

Jean-Romain, 2019-04-18:
@spono answer is correct but your columns must be named `X`, `Y`, `Z` (uppercase). If you want a more detailed and comprehensive answer please ask on https://gis.stackexchange.com/ that way I will take time to provide an answer that will be useful for other user as well. Please use github issues to report bugs.

closed 2019-04-18 by Jean-Romain