r-lidar/lidR #248: [2.1] lastransform doesn't find EPSG code

spono, 2019-04-26:
ciao JR,
I'm having some issues with `lastransform `and I don't understand if I'm missing/messing something.

**1st case**: using proj4strings
```
projection(las) = projectCRS
las = lastransform(las, new_projectCRS)

Error: EPSG code not found. Try to use the function epsg() manually.

# tested also with projectCRS = sp::CRS("+init=epsg:32632")
# and new_projectCRS as both text and CRSobj
```

**2nd case**: using epsg code 
```
projectCRS = 32632
epsg(las) = projectCRS
las = lastransform(las, new_projectCRS) # new_projectCRS as CRSobj

Error: EPSG code not found. Try to use the function epsg() manually.
```

Ideas?

all the best

(P.S. TBH, I didn't test it on 2.0.x: I tried only once in the real beginning and I remember a similar behaviour)

Jean-Romain, 2019-04-26:
It is a known bug related to epsg code search that should be fixed. 

In theory you can use the function `projection` to update the CRS of a LAS object. This updates the `proj4string` and retrieves the epsg code to update the header of the LAS object and ensure that the CRS will be written in the las file.

``` r
library(lidR)

crs  <- CRS("+proj=utm +zone=32 +datum=WGS84 +units=m +no_defs +ellps=WGS84 +towgs84=0,0,0")
r    <- raster()
projection(r) <- crs

LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
las <- readLAS(LASfile)
projection(las)
#> [1] "+proj=utm +zone=17 +datum=NAD83 +units=m +no_defs +ellps=GRS80 +towgs84=0,0,0"
epsg(las)
#> [1] 26917
projection(las) <- projection(r)
projection(las)
#> [1] "+proj=utm +zone=32 +datum=WGS84 +units=m +no_defs +ellps=WGS84 +towgs84=0,0,0"
epsg(las)
#> [1] 32632
```
But sometime the function `rgdal::showEPSG` fails at finding the epsg code of the proj4string. In your case the epsg code of `new_projectCRS` was not found. The crs transformation was ok but then, when trying to update the header with the epsg code, it failed.

This is not hard to fix. Please show me `new_projectCRS` this will help me to have an example of failing CRS

spono, 2019-04-28:
I'm willing to work in "Monte Mario Italy 1",

**EPSG:** 3003

**proj4:** "+proj=tmerc +lat_0=0 +lon_0=9 +k=0.9996 +x_0=1500000 +y_0=0 +ellps=intl +towgs84=-104.1,-49.1,-9.9,0.971,-2.917,0.714,-11.68 +units=m +no_defs"

Jean-Romain, 2019-04-29:
Fixed

- [X] Error became a warning. The worst case is now proj4string updated + epsg code removed in VLR
- [X] `projection` is able to parse the `+init:epsg=` string to find epsg if provided. Thus even is the proj4 does not enable to retrieve epsg it give a second chance to get the good one.

``` r
library(lidR)
#> Le chargement a nécessité le package : raster
#> Le chargement a nécessité le package : sp

LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
las = readLAS(LASfile)

projection(las)
#> [1] "+proj=utm +zone=17 +datum=NAD83 +units=m +no_defs +ellps=GRS80 +towgs84=0,0,0"
epsg(las)
#> [1] 26917

projection(las) <- CRS("+init=epsg:3003")

projection(las)
#> [1] "+proj=tmerc +lat_0=0 +lon_0=9 +k=0.9996 +x_0=1500000 +y_0=0 +ellps=intl +towgs84=-104.1,-49.1,-9.9,0.971,-2.917,0.714,-11.68 +units=m +no_defs"
epsg(las)
#> [1] 3003

projection(las) <- CRS("+proj=tmerc +lat_0=0 +lon_0=9 +k=0.9996 +x_0=1500000 +y_0=0 +ellps=intl +towgs84=-104.1,-49.1,-9.9,0.971,-2.917,0.714,-11.68 +units=m +no_defs")
#> Warning: EPSG code not found: header not updated. Try to use the function
#> epsg() manually to ensure CRS will be written in file.

projection(las)
#> [1] "+proj=tmerc +lat_0=0 +lon_0=9 +k=0.9996 +x_0=1500000 +y_0=0 +ellps=intl +towgs84=-104.1,-49.1,-9.9,0.971,-2.917,0.714,-11.68 +units=m +no_defs"
epsg(las)
#> [1] 0
```

<sup>Created on 2019-04-29 by the [reprex package](https://reprex.tidyverse.org) (v0.2.1)</sup>

commit daeb5e07 (2019-04-29): Fix #248 projection never fails
closed 2019-04-29 by Jean-Romain