r-lidar/lidR #249: readLAS fail (LASlib internal error)

jfbourdon, 2019-05-01:
I recently updated lidR and I'm now unable to read LAS/LAZ files that worked perfectly before.
I was before on lidR (2.0.0), rlas (1.2.8) and R (3.4.3) and updated to lidR (2.0.2), rlas (1.3.2) and R (3.6.0).

The error that readLAS returns:
`Error: LASlib internal error. See message above.
Error in C_reader(ifiles, ofile, select, filter, filter_wkt) :
  LASlib internal error. See message above.`

Jean-Romain, 2019-05-01:
Please show me which file are failing. In `rlas 1.3.2` `LASlib` has been entirely updated but I did not notice any file I was not able to read.

jfbourdon, 2019-05-01:
Here is a small LAZ ([link](https://transfert.mffp.gouv.qc.ca/?ShareToken=5ACCAACA2FD9687F20751DA652E87041D61B06E0
))

Jean-Romain, 2019-05-02:
I cannot reproduce. Does the following work for you?

```r
library(rlas)
read.las("~/Téléchargements/clip_LiDAR.laz")
read.lasheader("~/Téléchargements/clip_LiDAR.laz")
```