r-lidar/lidR #254: LAScatalog summary reporting area in kunits² when no CRS

jfbourdon, 2019-05-28:
If tiles within a catalog have no CRS, the area reported in the summary is given in _kunits²_ instead of _units²_ has expected.

```r
> catalog("/catalog/")
class       : LAScatalog
extent      : 202000 , 203000 , 5502000 , 5503000 (xmin, xmax, ymin, ymax)
coord. ref. : NA 
area        : 1 kunits²
points      : 8.59 million points
density     : 8.6 points/units²
num. files  : 1
```

Summary of a LAS object with an invalid CRS:
```r
> rlas::readLAS("/catalog/no_CRS.laz")
class        : LAS (LASF v1.2)
point format : 1
memory       : 655.7 Mb 
extent       :202000, 203000, 5502000, 5503000 (xmin, xmax, ymin, ymax)
coord. ref.  : NA 
area         : 999990 units² (convex hull)
points       : 8593501 points
density      : 8.59 points/units²
names        : X Y Z gpstime Intensity ReturnNumber NumberOfReturns ScanDirectionFlag EdgeOfFlightline Classification Synthetic_flag Keypoint_flag Withheld_flag ScanAngleRank UserData PointSourceID
```

jfbourdon, 2019-05-28:
I justed looked up the code in _LASmethods-LAScatalog.r_ and understand why it is written that way. Could a condition be added so that if no valid CRS is found, the prefix _thousand_ would be added instead of _k_?

Jean-Romain, 2019-05-28:
Ok that makes sense

wiesehahn, 2022-06-14:
I am not sure if this is expected to be changed in recent versions but I get `kunits²` with version 4.0.1

Jean-Romain, 2022-06-14:
I don't know either. Some mess somewhere.

closed 2022-06-30 by Jean-Romain