r-lidar/lidR #256: Error reading .las data, worked before updates

garmzy, 2019-06-05:
I recently updated some packages and I started getting an error when I run this simple code:

> LASfile3 <- ('D:/Courses/kpn/KPN project/cloud.las')
> las3 = lidR::readLAS(LASfile3, select = "*+")

Here is the error.

Error: 'readlasheader' is not an exported object from 'namespace:rlas'

What am I doing wrong?

Thank you!

Cory

closed 2019-06-05 by garmzy

Jean-Romain, 2019-06-06:
You probably updated the `rlas` package but not the `lidR` package. `readlasheader` is function replaced for a long time in `rlas`. You are likely to use a very old version of `lidR`. Please ensure you are using recent version of both `rlas` and `lidR`.

garmzy, 2019-06-06:
Thank you! This package is such a big help for my PhD research. You rock!!!