r-lidar/lidR #258: gpstime attribute format

komazsofi, 2019-06-08:
Dear Jean-Romain,

I have an issue with the number format of the gpstime attribute.

Instead of 78927560.001162528992 after reading we only have 78927560 in the gpstime attribute. My question is whether it is possible to read the gpstime in the original format? 

I am currently developing a script which able to detect dropout points on water (see https://onlinelibrary.wiley.com/doi/abs/10.1002/esp.1853) because that is really valuable information for me to detect exactly where is the land-water edge. For this, I need to detect the difference between gpstimes in order to detect the locations where the gpstime difference is higher then what we expect based on the acquisition parameter of ALS data, then we can interpolate in the missing points.  But with the current number format, this information is lost. 

Thank in advance for your help,
Zsofia

Jean-Romain, 2019-06-08:
Hi @komazsofi 

If gpstime is badly read it is an issue for the `rlas` package. But why do you think it is badly read? R usually don't print more than 8 digits. What if:

```r
options(digits = 20)
head(las$gpstime)
```

komazsofi, 2019-06-08:
That's it. I have been misled by printing out the values without setting the digits. Thank you.

closed 2019-06-08 by Jean-Romain