r-lidar/lidR #259: How to configure lidR catalog to save raster files?

komazsofi, 2019-06-10:
Hi J-R,

I have a small issue with configuring the extracted raster files with grid_metrics() function. I have tried to ask the question on StackOverflow (https://stackoverflow.com/questions/56504206/how-to-configure-lidr-catalog-to-save-raster-files) because I think this is not a bug it is more I did not read clear documentation about how to set the export. But there I did not get an answer (maybe I tagged wrongly the question) so I am asking here.

My question is how to set in the catalog options that:

1. be able to overwrite the results with a second run if I want to run my script several times and I do not want to name each output differently
2. instead of ".tif" I would like to get ".grd" files as output

My current code (which is working) is the following

´´´
normalizedctg=catalog(workingdirectory)
opt_output_files(normalizedctg) <- paste(workingdirectory,"/clipped/normalized/{XLEFT}_{YBOTTOM}_height",sep="")
height = grid_metrics(normalizedctg, max(Z),res = resolution)
´´´
But with a second run on the same code I get the following error:

´´´
Error in .getGDALtransient(x, filename = filename, options = options,  : 
  filename exists; use overwrite=TRUE
´´´

Jean-Romain, 2019-06-10:
Good question that have a not very well documented answer. You were right to ask on a forum instead of github but please ask on https://gis.stackexchange.com/. There is a tag `lidr` to ask question about `lidR` on gis.stackexchange and I get a notification. There are also other users who can anwser to questions. Use github to report bugs only. I do not anymore answer to question on github.

Jean-Romain, 2019-06-10:
Notice that your question has been moved to gis.stackechage. You don't need to post another one.

closed 2019-06-10 by Jean-Romain

Jean-Romain, 2019-06-11:
For future reader: answer [here](https://gis.stackexchange.com/questions/325367/how-to-configure-lidr-catalog-to-save-raster-files/325369#325369)