r-lidar/lidR #260: RGL displaying clouds very slow

spono, 2019-06-15:
Ciao JR,
in the last 2 months, the RGL device got ultra-slow (1-2 minutes) in showing even small pointclouds (10.000 pts). I tried updating, reinstalling, etc, but nothing changed.

I know that RGL is not your package but I wanted to ask you first because you might have a better idea of how the two packages interact.

Btw, I'm using the `devel` installation of lidR 2.1 (on Windows).

Thanks!

Jean-Romain, 2019-06-15:
If it was ok before, and slow after, you must figure out when is "before" and when is "after". But the code of `plot` did not change since v2.0.0.  You can try to display the point cloud outside the plot function. Roughtly `lidR` use `rgl` like the following:

```r
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
x    = lidR::readLAS(LASfile)
minx = min(x@data$X)
miny = min(x@data$Y)
with = list(x = x@data$X - minx, y = x@data$Y - miny, z = x@data$Z)

rgl::open3d()
do.call(rgl::points3d, with)
```

If the behavior is different from the package something unexpected arises in my `plot` function.

spono, 2019-06-15:
mmm...same behaviour, but running the single steps it seems that the issue arises with `rgl::open3d()` while `do.call(rgl::points3d, with)` is quick in plotting the single points

Jean-Romain, 2019-06-15:
So I can't help you more.

closed 2019-06-15 by Jean-Romain