r-lidar/lidR #261: las@data: adding variable without copy

floriandeboissieu, 2019-06-18:
There seem to be a problem with functions adding a variable to data without copying it first, e.g. laspulse. The new variable is under a new reference while the rest of the table has the same reference as the old table. Thus this last part of the table reacts at the reordering of the old table while the new variable does not.

Here is an example applied to laspulse (tested on R 3.4.4, lidR2.0.3, data.table 1.12.2):
```
library(lidR)
library(data.table)
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
las <- readLAS(LASfile)
las1 <- laspulse(las)
las2 <- LAS(copy(las@data), las@header, las@proj4string)
las2 <- laspulse(las2)

setorder(las@data, ReturnNumber)
# the order has also moved in las1@data but only for the variables in las@data, i.e. not for pulseID
all(las1@data$Intensity == las@data$Intensity)
las1@data[,.(pulseID)] # still starting by 1 and ending by 56979

setorder(las2@data, ReturnNumber)
all(las2@data$Intensity == las1@data$Intensity) # TRUE
all(las2@data$pulseID == las1@data$pulseID) # but FALSE...
```
I see several solutions to that:
- copy the las@data at the beginning of these functions
- come back to reference processing inside function
- specify it in the documentation, so that people make there own affectation by reference las <- laspulse(las) or there own copy before processing

Maybe you see a better solution?

Jean-Romain, 2019-06-18:
This is indeed a complex and unexpected behavior that comes from `data.table`. Working with reference is dangerous in R. This unexpected behavior may have important consequences when using successively two functions that order the points by reference such as `laspulse`.

```r
library(data.table)

dt1 = data.table(a = 1:10, b = 10:1)
dt1
dt2 = dt1
dt2[["c"]] <- 11:20
dt2

a1 = dt1[["a"]]
a2 = dt2[["a"]]
pryr::address(a1)
pryr::address(a2)

setorder(dt1, b)
dt1
dt2
```

> copy the las@data at the beginning of these functions

No way. I spend the half of my time finding tricks to avoid deep copy and useless memory allocation and you know how hard it is in R. Without that optimization the lidR package is almost unusable. Recently I made a mistake writing a code that includes an unexpected deep copy when processing big tiles and I ran out of memory while the corrected code ran smoothly.

> come back to reference processing inside function

No way. This is too complex for most user actually. And anyway I completely redesigned `lidR` in version 2.0 breaking former compatibility I cannot do it again.

> specify it in the documentation, so that people make there own affectation by reference las <- laspulse(las) or there own copy before processing

Well, people don't read enough the doc. And anyway this issue is complex and most users won't be able to understand the problem properly. I can document it more but it is actually partially documented in the [vignette](https://cran.r-project.org/web/packages/lidR/vignettes/lidR-LAS-class.html) (section deep copy and shallow copy). And yet I missed the problem myself...

The behavior you are pointing out is a major big problem. I can't see any good solution right now. I need to think about it.

floriandeboissieu, 2019-06-18:
Maybe you know about function .Internal(inspect()), but just in case 
not, it helped me find out about the problem:

https://stackoverflow.com/questions/10225098/understanding-exactly-when-a-data-table-is-a-reference-to-vs-a-copy-of-another

floriandeboissieu, 2019-06-18:
Maybe an argument to the function would also solve the pb, deep copy if 
copy=T (default), like it is if copy=F documenting the risk.

Jean-Romain, 2019-06-19:
Yes I know about `.Internal(inspect())` but I think that anything I can build from that will be an over-engineered and dangerous options. I can see two options.

- `copy = TRUE` is a simple a good option. I can see any advantage actually at setting `copy = TRUE` but at least it adds documentation everywhere it is needed.
- Restore the modification in place and return a reference:

```r
lasfunction = function(las)
{
  [...]
  las@data[, name := data]
  return(las)
}
```
This way

```r
las = lasfuntion(las) # works
lasfunction(las) # works the same
las2 = lasfunction(las) # updates las and las2 share the address of las
```

In the latest example the behavior might surprise many users but I can't see any serious consequences

closed 2020-02-23 by Jean-Romain