r-lidar/lidR #263: Best way to read in extra dim (laspy)

bw4sz, 2019-06-27:
I am working on releasing a benchmark dataset for trees in RGB + Lidar and I'd like the format to match the most common use cases and packages. I had been intended on proving a tree index using the user_data column. Alot of preprocessing happens in python and i'm writing from laspy. I realized that user_data is clamped 0-255, meaning that trees do not have a unique index. I'm starting to write an extra dimension function

https://stackoverflow.com/questions/50815580/appending-an-index-to-laspy-file-las

but looking here, it looks like lidR is going to ignore those columns

https://cran.r-project.org/web/packages/lidR/vignettes/lidR-LAS-class.html

and i'll need users to lasadddata anyways. Perhaps it would be best to just do a lookup table based on x,y,z? I'm worried about precision rounding among platforms. Thoughts welcome.

@brycefrank in case he wants to join in.

Jean-Romain, 2019-06-27:
If you have question about using the `lidR` package please ask on https://gis.stackexchange.com/ with the tag `lidr`. Me or somebody else will be able to help you and your question will benefit to other users. I no longer answer to question on github because it takes a lot of time and do not benefit to other users (github issues are poorly indexed by search engines). Please only report bugs on github. Regards.

closed 2019-06-27 by Jean-Romain

bw4sz, 2019-06-27:
totally.

bw4sz, 2019-06-27:
for those who end up here: https://gis.stackexchange.com/questions/327146/best-way-to-maintain-extra-data-for-laz-for-use-in-r-and-python

Jean-Romain, 2019-06-27:
If you can transform your question into a "this is a las file with extra attribute written with laspy but lidR does not read it" issue it might be a bug report or an enhancement suggestion for the [rlas](https://github.com/Jean-Romain/rlas) pakage

bw4sz, 2019-06-27:
Thanks for your help. It was the datatypes.