r-lidar/lidR #268: problem with class inheritance

rsbivand, 2019-08-04:
I'm seeing: 

```
Browse[2]> class(res)
[1] "LAScatalog"
attr(,"package")
[1] "lidR"
...
Browse[3]> contour <- rgeos::gUnaryUnion(as(catalog, "SpatialPolygonsDataFrame"))
Browse[3]> 
debug: contour <- rgeos::gUnaryUnion(catalog)
Browse[3]> 
Error in RGEOSUnaryPredFunc(spgeom, byid, "rgeos_isvalid") : 
  rgeos_convert_R2geos: invalid R class, unable to convert
```

in `lidR::readLAScatalog()`. 

I'm making changes to `rgeos::gUnaryUnion()` (on R-Forge https://r-forge.r-project.org/projects/rgeos/, rev 604) to try to handle the apparently stricter handing of invalid geometries in GEOS 3.7.2 than in earlier GEOS releases. This means that I need to check their validity. I can see that LAScatalog inherits from "Spatial", but it may be prudent to coerce objects in **lidR**. It may be that with earlier GEOS, and no call in gUnaryUnion() to gIsValid(), things worked, but as current and future GEOS are stricter, gIsValid() is needed.

Alternatively, you can edit lidR::is.overlapping() adding `checkValidity=0L` to the gUnaryUnion() call, to return to the state for GEOS < 3.7.2 and rgeos < 0.5 (the settings for checkValidity= change in rgeos >= 0.5).

Jean-Romain, 2019-08-05:
A `LAScatalog` inherits a `SpatialPolygonsDataFrame` but it is not a problem to coerce it to coerse it to strict `SpatialPolygonsDataFrame` before to enter `g*` functions.

closed 2019-08-05 by Jean-Romain

rsbivand, 2019-08-05:
Thanks - the C function used to convert objects to the GEOS representation only scans for **sp** classes. **rgeos** 0.5-1 is on CRAN now, so the errors will start to show in **lidR** check results.

Jean-Romain, 2019-08-05:
I'll make an update asap. Thanks

beedawg1985, 2020-04-27:
I'm still getting this error — just wanted to check that the workaround is to still edit the functions as described above?

Jean-Romain, 2020-04-27:
Please show a reproducible example.

marcobaldoroma, 2020-04-27:
Sorry I didn't understand the error?

Ottieni Outlook per Android<https://aka.ms/ghei36>

marcobaldoroma, 2020-04-27:
About LidR?

Ottieni Outlook per Android<https://aka.ms/ghei36>

marcobaldoroma, 2020-04-27:
I'm not start to work on thia code.

Ottieni Outlook per Android<https://aka.ms/ghei36>

________________________________
From: marco baldo <baldomarco@live.it>
Sent: Monday, April 27, 2020 5:38:20 PM
To: Jean-Romain/lidR <reply@reply.github.com>
Subject: Re: [Jean-Romain/lidR] problem with class inheritance (#268)

About LidR?

Ottieni Outlook per Android<https://aka.ms/ghei36>

marcobaldoroma, 2020-04-27:
I didn't start sorrym

Ottieni Outlook per Android<https://aka.ms/ghei36>

________________________________
From: marco baldo <baldomarco@live.it>
Sent: Monday, April 27, 2020 5:39:06 PM
To: Jean-Romain/lidR <reply@reply.github.com>
Subject: Re: [Jean-Romain/lidR] problem with class inheritance (#268)

I'm not start to work on thia code.

Ottieni Outlook per Android<https://aka.ms/ghei36>

________________________________
From: marco baldo <baldomarco@live.it>
Sent: Monday, April 27, 2020 5:38:20 PM
To: Jean-Romain/lidR <reply@reply.github.com>
Subject: Re: [Jean-Romain/lidR] problem with class inheritance (#268)

About LidR?

Ottieni Outlook per Android<https://aka.ms/ghei36>

beedawg1985, 2020-04-27:
```
R version 3.6.0 (2019-04-26)
Platform: x86_64-apple-darwin15.6.0 (64-bit)
Running under: macOS  10.15.4

Matrix products: default
BLAS:   /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libBLAS.dylib
LAPACK: /Library/Frameworks/R.framework/Versions/3.6/Resources/lib/libRlapack.dylib

locale:
[1] en_GB.UTF-8/en_GB.UTF-8/en_GB.UTF-8/C/en_GB.UTF-8/en_GB.UTF-8

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] lidR_2.1.0        raster_3.1-5      sp_1.4-1          RPostgreSQL_0.6-2
[5] DBI_1.1.0        

loaded via a namespace (and not attached):
 [1] Rcpp_1.0.4.6       lattice_0.20-38    codetools_0.2-16   class_7.3-15      
 [5] sf_0.9-2           packrat_0.5.0-10   grid_3.6.0         magrittr_1.5      
 [9] units_0.6-6        e1071_1.7-3        KernSmooth_2.23-15 lazyeval_0.2.2    
[13] data.table_1.12.8  tools_3.6.0        compiler_3.6.0     rlas_1.3.5        
[17] classInt_0.4-3     rgeos_0.5-2 
```

```
> lazFolder <- 'lidar/england-laz-2014_3520198/su'
>   a <- readLAScatalog(lazFolder)
Error in RGEOSUnaryPredFunc(spgeom, byid, "rgeos_isvalid") : 
  rgeos_convert_R2geos: invalid R class LAScatalog, unable to convert
> 
```
The folder contains various .laz and .txt files downloaded from:
https://environment.data.gov.uk/DefraDataDownload/?Mode=survey

Was able to read the catalog by editting the function as described above.

```
> trace(is.overlapping, edit=T)
Tracing function "is.overlapping" in package "lidR"
[1] "is.overlapping"
>   a <- readLAScatalog(lazFolder)
Be careful, some tiles seem to overlap each other. lidR may return incorrect outputs with edge artifacts when processing this catalog.
```

marcobaldoroma, 2020-04-27:
Ok

Ottieni Outlook per Android<https://aka.ms/ghei36>

https://environment.data.gov.uk/DefraDataDownload/?Mode=survey

Jean-Romain, 2020-04-27:
Update `lidR`. You are using `v2.1.0` which is 8 month old.

marcobaldoroma, 2020-04-27:
Okay Thank you a lot, are you mean on R right?!

Ottieni Outlook per Android<https://aka.ms/ghei36>

beedawg1985, 2020-04-28:
> Update `lidR`. You are using `v2.1.0` which is 8 month old.

Thanks and apologies for not checking.

commit 7cf99208 (2019-08-05): coerce to SpatailPolygonsDataFrame (fix #268)