r-lidar/lidR #271: [2.1.3] 'ORIGINALFILENAME' not found

spono, 2019-09-03:
Ciao JR,
running the following simple `catalog `analysis
```
ctg = catalog(file.path(wdir,'2011','LAS'))
opt_chunk_buffer(ctg) = 10
opt_chunk_size(ctg) = 0 # by file
opt_output_files(ctg) = paste0(file.path(wdir,'2011','LAS','processed'), "/{ORIGINALFILENAME}")
opt_laz_compression(ctg) = TRUE

tmp = catalog_apply(ctg, grfun)
```

I get the following error:
```
Error in eval(parse(text = text, keep.source = FALSE), envir) : 
  object 'ORIGINALFILENAME' not found
```

I tested it on two different datasets and the result is the same.
Can you confirm it?

Jean-Romain, 2019-09-03:
Could you please make a reproducible example ? This works for me:

```r
library(lidR)
grfun = function(cl) {
  las = readLAS(cl)
  if (is.empty(las)) return(NULL)
  las = lasfilterdecimate(las, random(0.25))
  return(las)
}

ctg = catalog("~/Documents/ALS data/Haliburton dataset/Landscape LiDAR/")
ctg = ctg[1:4,]
opt_chunk_buffer(ctg) = 10
opt_chunk_size(ctg) = 0 # by file
opt_output_files(ctg) = paste0(tempdir(), "/{ORIGINALFILENAME}")
opt_laz_compression(ctg) = TRUE

tmp = catalog_apply(ctg, grfun)
tmp

#> [[1]]
#> [1] "/tmp/Rtmp6OOS09/CN_679_5016.laz"
#> 
#> [[2]]
#> [1] "/tmp/Rtmp6OOS09/CN_679_5017.laz"
#> 
#> [[3]]
#> [1] "/tmp/Rtmp6OOS09/CN_680_5011.laz"
#> 
#> [[4]]
#> [1] "/tmp/Rtmp6OOS09/CN_680_5012.laz"
```

closed 2019-09-10 by Jean-Romain

spono, 2020-02-05:
Ciao JR,
I reopen this weird (...as all my issues...) bug.
This time [v 2.2.2] I'm trying to normalise a file using the `catalog `approach 
```
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
ctg = readLAScatalog(LASfile)
opt_chunk_size(ctg) <- 150
opt_chunk_buffer(ctg) <- 10
opt_output_files(ctg) <- file.path(las.folder,'norm','*_norm')
opt_laz_compression(ctg) <- TRUE
  
ctg = lasnormalize(ctg, tin())
```
but (independently from the file [mine or Megaplot]) I still get the error:
```
Error in eval(parse(text = text, keep.source = FALSE), envir) : 
  object 'ORIGINALFILENAME' not found
```

I tested also with the long form `{ORIGINALFILENAME}` or the alternative used in #309  (i.e. `{*}`) but I get the same result.

Jean-Romain, 2020-02-05:
You made chunks of 150 x 150. The original file names does not have meaning here since you are not processing by file. So `ORIGINALFILENAME` is not defined. It is defined only if  `opt_chunk_size(ctg) <- 0`. I know this case should throw a more informative warning.

spono, 2020-02-05:
oh, ok, makes sense.
thanks for clarification!

Jean-Romain, 2020-02-05:
Your example now throws informative message

``` r
library(lidR)
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
ctg = readLAScatalog(LASfile)
opt_chunk_size(ctg) <- 150
#> Be careful, a chunk size smaller than 250 is likely to be irrelevant.
opt_chunk_buffer(ctg) <- 10
opt_output_files(ctg) <- file.path(tempdir(),'norm','*_norm')
#> ORIGINALFILENAME template has been used but the chunk size is not 0. This template makes sense only when processing by file.
opt_laz_compression(ctg) <- TRUE

ctg = lasnormalize(ctg, tin())
#> Error: The template {ORIGINALFILENAME} makes sense only when processing by file (chunk size = 0). It is undefined otherwise.
```

<sup>Created on 2020-02-05 by the [reprex package](https://reprex.tidyverse.org) (v0.3.0)</sup>

azh2, 2020-03-03:
Thank you for pointing this out.