r-lidar/lidR #272: Spatial information changes when writing and reading las with no native crs

BBranoff, 2019-09-03:
I've noticed this issue only for las objects which are read with no native crs information, so clearly an issue with the data. However, if the crs can be obtained and set to the las object, spatial information for that object is incorrect only after writing the transformed las...

'
###  download LiDAR file with native NA crs in lidR
Lurl <- "ftp://rockyftp.cr.usgs.gov/vdelivery/Datasets/Staged/Elevation/LPC/Projects/USGS_LPC_TN_27County_blk4_2015_LAS_2018/laz/USGS_LPC_TN_27County_blk4_2015_2038301SE_LAS_2018.laz" 
temp <- tempfile(fileext = ".laz")
Lfile <-  download.file(Lurl,temp)
L <- lidR::readLAS(temp)
### check area and crs
area(L)
crs(L)
L@proj4string

##   get the intended crs from the tile boundaries metadata
bboxURL <- "ftp://rockyftp.cr.usgs.gov/vdelivery/Datasets/Staged/Elevation/LPC/FullExtentSpatialMetadata/FESM_LPC_Proj.gpkg"
bboxTemp <- tempfile(fileext = ".gpkg")
bboxFile <- download.file(bboxURL,bboxTemp)
bbox <- read_sf(bboxTemp)
###  the project_id for this LiDAR data is in the Lurl above ("27County_blk4_2015")
crs <- bbox$refsys[grepl("27County_blk4_2015",bbox$project_id)]
crs
##  this is epsg 6576
##  set the crs of the las to the intended crs
projection(L) <- st_crs(6576)$proj4string
## check the area 
area(L)
###  can also transform the las, but does not change the problem
L <- lastransform(L,crs(L))

###  transform the las to the utm zone in meters
LUTM <- lastransform(L,"+proj=utm +zone=16 +ellps=WGS84 +datum=WGS84 +units=m +no_defs")
## check the area, should be about 0.3048*0.3048 less than the original area in ft2 (1m=0.3048 ft)
area(LUTM)
### can also change the Z data to meters, but does not change the problem
LUTM@data$Z <- LUTM@data$Z*0.3048
## check las summary
LUTM
### seems fine, so save
writeLAS(LUTM,sub(".laz","2.las",temp))

##  open the same file and check the area, I get 0
##  also notice the extent changes when writing and opening the file
##  lastransform invokes lasupdateheader(), so supposedly the header has already been updated
LUTM2 <- readLAS(sub(".laz","2.las",temp))
area(LUTM2)
LUTM2
'

BBranoff, 2019-09-03:
``

Jean-Romain, 2019-09-03:
Could you please explain why you open and immediately close the issue. Is it ok for you? Could you explain more carefully the issue? What is wrong at the end?

BBranoff, 2019-09-03:
Sorry, accidentally closed the issue while editing the original issue for clarity.

Jean-Romain, 2019-09-03:
Ok, so i'll try your example

Jean-Romain, 2019-09-03:
`Y` seems to be completely corrupted.

```r
> LUTM@data
               X       Y        Z
     1: 614617.9 3892840 527.9837
     2: 614715.2 3892841 528.4775
     3: 614782.7 3892842 549.9384
     4: 614848.1 3892843 533.0038
     5: 614853.0 3892843 533.0038

> LUTM2@data
               X        Y       Z
     1: 614617.9 -2147484 527.984
     2: 614715.2 -2147484 528.477
     3: 614782.7 -2147484 549.938
     4: 614848.1 -2147484 533.004
     5: 614853.0 -2147484 533.004
```

This is due to an integer overflow. You can fix it by setting manually an offset into the header

```r
LUTM@header@PHB$`X offset` <- min(LUTM$X)
LUTM@header@PHB$`Y offset` <- min(LUTM$Y)
writeLAS(LUTM,sub(".laz","2.las",temp))
LUTM2 <- readLAS(sub(".laz","2.las",temp))
```

```
> LUTM@data
               X       Y        Z
     1: 614617.9 3892840 527.9837
     2: 614715.2 3892841 528.4775
     3: 614782.7 3892842 549.9384
     4: 614848.1 3892843 533.0038
     5: 614853.0 3892843 533.0038

> LUTM2@data
               X       Y       Z
     1: 614617.9 3892840 527.984
     2: 614715.2 3892841 528.477
     3: 614782.7 3892842 549.938
     4: 614848.1 3892843 533.004
     5: 614853.0 3892843 533.004
```

This should be managed by `lastransform()`

Jean-Romain, 2019-09-04:
Fixed

```r
library(lidR)

Lurl  <- "ftp://rockyftp.cr.usgs.gov/vdelivery/Datasets/Staged/Elevation/LPC/Projects/USGS_LPC_TN_27County_blk4_2015_LAS_2018/laz/USGS_LPC_TN_27County_blk4_2015_2038301SE_LAS_2018.laz" 
temp  <- tempfile(fileext = ".laz")
Lfile <-  download.file(Lurl, temp)
L     <- lidR::readLAS(temp, "xyz", filter = "-thin_with_grid 6")
projection(L) <- CRS("+init=epsg:6576")

LUTM <- lastransform(L,"+proj=utm +zone=16 +ellps=WGS84 +datum=WGS84 +units=m +no_defs")
LUTM@data[, Z := Z*0.3048]

temp2 <- tempfile(fileext = ".laz")
writeLAS(LUTM, temp2)
LUTM2 <- readLAS(temp2)
plot(LUTM2)
```

closed 2019-09-04 by Jean-Romain

BBranoff, 2019-09-04:
Excellent

commit 35a2212c (2019-09-04): Fix #272