r-lidar/lidR #276: New function `point_metrics`

Jean-Romain, 2019-09-14:
Similary to `grid_metrics` that can apply any user-defined function by aggregating the points at the cell level it might be useful to add a function able to apply any user-defined function at the point level using the knn.

Actually such function exists internally in C++. This is what `lasdetectshape()` or `lasnormalize()` does. But these function have two different hard coded behaiors. We need something more flexible and more generic.

```r
m = point_metrics(las, mean(Z), k = 8)
```

bi0m3trics, 2019-09-14:
Just a thought, and of course this would depend on the shape of the cell but I seem to recall this also being in the c++ code, it would be nice to have an argument to specify a neighborhood distance or cell size to calculate the point metrics. I could see this being used for all kinds of things!

Jean-Romain, 2019-09-14:
I don't know if I will enable a radius search option. Radius search may contains a single point while knn always contains k points. It is a matter of memory optimization and bug safety mainly. I'll consider the option anyway.

bi0m3trics, 2019-09-14:
Understood... it was just a thought. The functionality is Shapes.cpp/.h anyway... ;)

Jean-Romain, 2019-09-16:
I'm done with a first draft of the function. I'm now exploring the R's C API to find even more optimizations.  However sad new, no radius queries. It is not possible (at least beyond my capabilities). It is more complex than using `Shape.h` :wink: 

With knn I can allocate a table with k rows only once and recycle the allocated memory for each points. With a radius search, the number of neighbors is variable meaning I cannot recycle the memory. In the latter case, the performances are terrible.

bi0m3trics, 2019-09-16:
Boo! Thanks for looking into it anyway.

Jean-Romain, 2019-09-16:
I'm really proud of what have been done the past 3 days. I made a very fast version of `point_metrics()`, a feature I needed myself for a long time but that was hard to build efficiently to me.

It uses R's C API stuff similarly to the `data.table` package. However I'm not as good as `data.table` team so it can contains unexpected bugs. I'm especially expecting bugs with objects not found in scope.  This feature requires a lot of production tests. If you want to build tools out of it please give me a feed back.

The documentation shows how to rebuild `lasdetectshape()` with `point_metrics()`. Astonishingly it is only 3 times slower. Indeed `lasdetectshape()` is pure C++ while `point_metrics` maps a user-defined function. There are much more overheads in the later. Also `point_metrics` cannot be parallelized safely.

To finish the function is not done yet. I will add an options `xyz = TRUE` to return the coordinates. I will try to find some other clever optimizations. And I'm not sure of the output format yet.

```r
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
las = readLAS(LASfile)

Rcpp::sourceCpp(code = "
#include <RcppArmadillo.h>
// [[Rcpp::depends(RcppArmadillo)]]

// [[Rcpp::export]]
SEXP eigen_values(arma::mat A) {  
arma::mat coeff;
arma::mat score;
arma::vec latent;
arma::princomp(coeff, score, latent, A);
return(Rcpp::wrap(latent));
}")

plane_metrics = function(x,y,z, th1 = 25, th2 = 6) {
  xyz <- cbind(x,y,z) 
  eigen_m <- eigen_values(xyz)
  is_planar <- eigen_m[2] > (th1*eigen_m[3]) && (th2*eigen_m[2]) > eigen_m[1]
  return(list(planar = is_planar))
}

system.time(M <- point_metrics(las, ~plane_metrics(X,Y,Z), k = 8))
#> 1.9 s

set_lidr_threads(1)
system.time(las <- lasdetectshape(las, shp_plane(), "planar"))
#>  0.7 s

all.equal(M$planar, las$planar)
#> [1] TRUE
```

TedwardErker, 2019-10-18:
Thanks!  I used the function to help separate out buildings from trees.  See the work here:

https://github.com/TedwardErker/lidR_urban_trees

Jean-Romain, 2019-10-18:
I designed the function exactly for this kind of usage. I'm glad you can take advantage of it. :+1:

closed 2019-11-11 by Jean-Romain

Jean-Romain, 2020-01-21:
> , it would be nice to have an argument to specify a neighborhood distance or cell size (as an alternative to k maybe) to calculate the point metrics. I could see this being used for all kinds of things!

radius (sphere) search is coming

Jean-Romain, 2020-01-27:
Radius search available in v2.3.0.