r-lidar/lidR #278: Parallelism

whit0348, 2019-09-14:
It appears that the addition of parallelism has no effect in processes like grid_metrics and lasnormalize while operating on catalog objects. It would be nice to have opt_cores back until parallelism is ironed out.

Jean-Romain, 2019-09-14:
`grid_metrics()` is not parallelized. Some algorithms of `lasnormalize()` are. `knnidw()` is parallelized, `tin()` is partially parallelized. Anyway the LAScatalog engine still supports chunk parallelization the same way than before. The function `opt_cores` has no more any effect but the feature is still here and better implemented. See the [changelog of version 2.1.0](https://github.com/Jean-Romain/lidR/blob/master/NEWS.md#lidr-v210-release-date-2019-07-13). And also the documentation relative to the parallelism `?lidR-parallelism`

whit0348, 2019-09-14:
Right, but now since opt_cores is gone, calculating grid_metrics on a catalog can take substantially longer than before the implementation of parallelism. This is because opt_cores essentially meant 'number of back-end instances of R to split the operation by', which now only 1 is allowed. It would be nice to give the opt_cores option for at least grid_metrics and the algorithms of lasnormalize that do not yet have parallelism.

Jean-Romain, 2019-09-14:
I recommend you to read the vignettes of the `future` package for an better understanding. But, briefly, nothing changed. The folllwing is strictly equivalent to the former `opt_cores(ctg) <- 4L`. 

```r
library(lidR)
library(future)
plan(multisession, workers = 4L)
ctg <- catalog("folder/")
hmean <- grid_metrics(ctg, mean(Z))
```

Formerly `plan()` was called internally which was not a good practice. Now it is called externally by the user everything is possible including remote computation on multiple machine (see fixes in [changelog v2.1.3](https://github.com/Jean-Romain/lidR/blob/master/NEWS.md#lidr-v213-release-date-2019-09-10)

closed 2019-09-18 by Jean-Romain