r-lidar/lidR #279: `catalog_laxindex` won't write when `opt_output_files` set on catalog

lucas-johnson, 2019-09-19:
I believe this is a bug. Or maybe I should be writing the .lax files to the originally directory? I wonder then, will the new normalized files need to be re-indexed? Thank you for taking a look at this. 

```
> opt_output_files(ctg) <- '/temp/normalized_{ORIGINALFILENAME}'
> lidR:::catalog_laxindex(ctg)
  |=====|   4%
Error: Trying to write an object of class numeric but this type is not supported.

> opt_output_files(ctg) <- ''
> lidR:::catalog_laxindex(ctg)
  |=============================================================================================================================| 100%
```

Jean-Romain, 2019-09-19:
This is a bug. Not a big one actually. `catalog_index()` writes `lax` file along with the ` las/z` files with the same names than original file names. There is no custom behavior. It should have warned about meaningless options but instead it failed badly.

lucas-johnson, 2019-09-19:
No worries 👍

commit f83a19d2 (2019-09-19): Fix #279
closed 2019-09-19 by Jean-Romain