r-lidar/lidR #280: noise removal

YaquanChang, 2019-09-20:
Hi Dr Jean-Romain,

Thank you for your amazing package. I found it seems that there is no noise removal function in the package. So far I could only remove the noise below the ground after normalization. But some noise above the ground could not be removed. Could you add a function for noise removal (e.g., removing noise over 3 or 5 standard deviations)?

Best Regards,
Yaquan

Jean-Romain, 2019-09-20:
Hi,

Sadly I won't add any noise removal function for the moment. The reason it that it does not exist any peer-reviewed method to achieve such task to my knowledge. `lidR` relies, as far as possible, on published algorithm with references to peer-reviewed journal articles. I you have any reference to provide I may consider the feasibility. Otherwise you must write this feature yourself from existing tools. [This vignettes](https://cran.r-project.org/web/packages/lidR/vignettes/lidR-catalog-apply-examples.html) shows an example of a user-defined `lasfilternoise` function that is similar to your suggestion.

YaquanChang, 2019-09-20:
Thank you for your suggestion!

closed 2019-09-20 by Jean-Romain