r-lidar/lidR #282: Tests of v2.2.0

Jean-Romain, 2019-09-25:
To @bi0m3trics @jfbourdon @ptompalski @tiagodc @spono @floriandeboissieu and to other users who are seeing that and who would like to test new features.

The version 2.2.0 will be released in January 2020. So it is not in a hurry. v2.2.0 will bring new features (not necessarily done yet) but also two very important internal changes that are already done. These changes should be invisible for the users **as long as it works without bugs**. 

Here you can help. The changes imply very deep modifications of the C++ code. It has been tested but it is prone to unexpected behaviors especially with weird point-clouds. I cannot test every single case of partially valid point-clouds. 

So if you want to test v2.2.0 please give me some feed back. Below an in-depth description of theses two changes.

--------

On overall the package is expected to be 2 to 3 times faster in many functions. This has been achieved by modifying the spatial indexation and the Delaunay triangulation

### Spatial Indexation

The package used a quadtree to make spatial queries (such as knn in `lasdetectshape()`, points in disc in `lmf()`, points in triangle in `tin()` or `pitfree()` and so on).

The quadtree as been replaced by a grid partition. The queries are twice faster meaning that some functions are twice faster such as `lmf()`, `wing2015()`, `lasdetectshape()` and many functions should be boosted.

I'm not expecting a lot of trouble from this change. But if you find weird outputs please report. Functions that relies on a spatial index are `lmf()` `wing2015()` `lasdetectshape()` `tin()` `dsmtin()` `pitfree()` `lassmooth()` `pmf()` `point_metrics()`

### Delaunay triangulation

The package used the `geometry` package to compute Delaunay triangulation. It was not as fast as I wanted and it consumed a lot of memory because of data type cast internally required. I wrote a Delaunay triangulation using `boost`. It is 3 times faster meaning than `tin()`, `dsmtin()` and `pitfree()` are 3 times faster and it consumes less memory.

However **I'm expecting a lot of trouble from this change**. The triangulation is computed using **integral** numbers but point cloud coordinates are **decimal** number. This is not a big deal since `las` files actually store integral coordinates that are converted to decimal with a scale factor (for a centimeter or millimeter accuracy) and an offset. So internally the coordinates are unscaled and unoffseted, triangulated and scaled and offseted back.

It works! But there are many cases where it cannot work.

1. **If the scales on x and y are different**: if _x_ have a millimeters precision and _y_ have a centimeter precision it is meaningless to triangulate their integral representation. This should never happen but in practice many `las` files are invalid so I won't be surprised to find such case. This case is handled internally and fall back to the former behavior using `geometry`

   ```r
   LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
   las = readLAS(LASfile)
   las@header@PHB[["Y scale factor"]] <- 0.001
   dtm = grid_terrain(las,1, tin())
   #> The delaunay triangulation fall back to the old slow method.
   #> This probably means that the xy scale factors are different and that the fast method cannot be applied.
    ```

2. **User modified manually the scale factors and the offsets**: This may happen because `readLAS()` throw a warning for invalid scale factor (see [this question on gis stackexchange](https://gis.stackexchange.com/questions/333324/rescaling-las-file-with-lidr-package-in-r)). This case is handled internally a throw an error.

    ```r
    LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
    las = readLAS(LASfile)
    las@header@PHB[["X scale factor"]] <- 0.012
    las@header@PHB[["Y scale factor"]] <- 0.012
    dtm = grid_terrain(las,1, tin())
    #>  Erreur : Internal error in C_interpolate_delaunay: xy coordinates were not converted to integer. Scale factors are likely to be invalid.
    ```

3. **A user creates a LAS object out of external data**: reading a csv file containing TLS data or whatever. This fails for the same reason. However it is handled internally because if a user wants to create a valid LAS object , the function `LAS()` should be used

    ```r
    data = data.table(X = runif(10), Y = runif(10), Z = runif(10), Classification = 2L)
    LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
    las = readLAS(LASfile)
    las@data <- data
    dtm = grid_terrain(las,1, tin())
    #>  Erreur : Internal error in C_interpolate_delaunay: xy coordinates were not converted to integer. Scale factors are likely to be invalid.

    las = LAS(data)
    #> Creation of a LAS object from data but without a header. Scale factors were set to 0.01 and XYZ coordinates were rounded to fit the scale factors.
    ```

If you have read so far, you might guess that there are probably some cases that are not handled internally yet. I'm sure the Delaunay triangulation will bring more trouble...

tiagodc, 2019-09-25:
Great Jean, as always!

I'll be glad to be an early tester of the new features, count me in.

In my case, for TLS data I often have to change the XYZ scale factors
manually, typically to 0.0001, so I'm falling in one of those triangulation
bug categories. I'll have a close look at the issues that might happen and
report them as I use the version.

Thanks for the heads up.

Best regards,
Tiago

Jean-Romain, 2019-09-25:
If your coordinates are in meter 0.0001 means a tenth of millimeter precision. It is likely to be overkilled imo. Anyway if you want to change the scale of a `LAS` object you can use `lasrescale()` and `lasreoffset()`. If you want to create a `LAS` object out of external data with an accuracy of 0.0001 this is not possible yet because `LAS()` will set an accuracy of 0.01 by default. But I can easily add an option change the scale if needed.

floriandeboissieu, 2019-09-26:
Nice! I'll make a try and tell you about it.

Jean-Romain, 2019-09-26:
Side note: the errors in the Delaunay triangulation are only here temporarily. Ideally they will be caught earlier to allows to fall back in the former method using `geometry`. But for now I prefer to fail explicitly to catch unexpected bugs.

closed 2019-11-22 by Jean-Romain