r-lidar/lidR #284: What las / laz file is used in the documentation example? Question / Not an issue

ns-1m, 2019-10-01:
Hi,

What las / laz file is used in the documentation example?

I tried this one below. It is not working.

Noli
~~~~~~~~`
> source('~/Documents/Work/lidR/lidR_1.R', echo=TRUE)

> setwd("~/Documents/Work/lidR")

> library(rlas)

> library(lidR)

> las <- readLAS("./extdata/MixedConifer.laz")

> #plot(las)
> 
> # Khosravipour et al. pitfree algorithm
> thr <- c(0,2,5,10,15)

> edg <- c(0, 1.5)

> chm <- grid_canopy(las, 1, pitfree(thr, edg))

> plot(chm)
Error in graphics::par(plt = bigplot) : 
  invalid value specified for graphical parameter "plt"
>

Jean-Romain, 2019-10-01:
Please be more specific. What is your question here? What is wrong with the example in the documentation? The example file are stored with the package. It depends on your system. On my computer it is stored here

```r
system.file("extdata", "Megaplot.laz", package="lidR")
#> [1] "/home/jr/R/x86_64-pc-linux-gnu-library/3.6/lidR/extdata/Megaplot.laz"
```

ns-1m, 2019-10-01:
Change the las file into Megaplot.laz still the same error.

~~~~~~`
> source('~/Documents/Work/lidR/lidR_1.R', echo=TRUE)

> setwd("~/Documents/Work/lidR")

> library(rlas)

> library(lidR)

> las <- readLAS("./extdata/Megaplot.laz")

> #plot(las)
> 
> # Khosravipour et al. pitfree algorithm
> thr <- c(0,2,5,10,15)

> edg <- c(0, 1.5)

> chm <- grid_canopy(las, 1, pitfree(thr, edg))

> plot(chm)
Error in graphics::par(plt = bigplot) : 
  invalid value specified for graphical parameter "plt"
>

bi0m3trics, 2019-10-01:
This looks like it might be a package conflict to me... what happens when you replace that last line with raster::plot (chm)?

ns-1m, 2019-10-01:
I think something wrong with the plot now.
~~~~~~~~~

> setwd("~/Documents/Work/lidR")

> library(rlas)

> library(lidR)

> las1 <- readLAS("./extdata/drawno.laz")

> #plot(las)
> 
> 
> #library(lidR)
> #las = readLAS("drawno.laz")
> las2 = lasnormalize(las1, tin())

> plot(las2)

> chm1 = grid_canopy(las2, 0.5, p2r())

> plot(chm1, col = height.colors(50))
Error in graphics::par(plt = bigplot) : 
  invalid value specified for graphical parameter "plt"
In addition: Warning message:
There were 3 degenerated ground points. Some X Y coordinates were repeated but with different

ns-1m, 2019-10-01:
OK,  raster::plot(....). I got this error as well.

~~~~~~
> source('~/Documents/Work/lidR/lidR_2.R', echo=TRUE)

> setwd("~/Documents/Work/lidR")

> library(rlas)

> library(lidR)

> las1 <- readLAS("./extdata/drawno.laz")

> #plot(las)
> 
> 
> #library(lidR)
> #las = readLAS("drawno.laz")
> las2 = lasnormalize(las1, tin())

> plot(las2)

> chm1 = grid_canopy(las2, 0.5, p2r())

> raster::plot(chm1, col = height.colors(50))
Error in graphics::par(plt = bigplot) : 
  invalid value specified for graphical parameter "plt"
In addition: Warning message:
There were 3 degenerated ground points. Some X Y coordinates were repeated but with different Z coordinates. min Z were retained. 
>

ns-1m, 2019-10-01:
I got the same error in Linux but the error message is better in Linux and the solution is:

par(mar=c(1.1,1,1))

closed 2019-10-01 by ns-1m