r-lidar/lidR #286: Handle more error in pitfree()

Jean-Romain, 2019-10-09:
All the following cases should be handled internally and should fail nicely

``` r
library(lidR)
#> Le chargement a nécessité le package : raster
#> Le chargement a nécessité le package : sp

LASfile <- system.file("extdata", "MixedConifer.laz", package="lidR")
las <- readLAS(LASfile)

# 0 points
las2 = lasfilter(las, Z >= 33)
chm <- grid_canopy(las2, res = 0.5, pitfree(thresholds = c(0,2,5,10,15,20)))
#> Warning in min(x@data$X): aucun argument trouvé pour min ; Inf est renvoyé
#> Warning in max(x@data$X): aucun argument pour max ; -Inf est renvoyé
#> Warning in min(x@data$Y): aucun argument trouvé pour min ; Inf est renvoyé
#> Warning in max(x@data$Y): aucun argument pour max ; -Inf est renvoyé
#> Error in validityMethod(object): invalid extent: xmin >= xmax

# 0 points
las2 = lasfilter(las, Z >= 33)
chm <- grid_canopy(las2, res = 0.5, pitfree(thresholds = c(0,2,5,10,15,20),  subcircle = 0.2))
#> Warning in min(x@data$X): aucun argument trouvé pour min ; Inf est renvoyé
#> Warning in max(x@data$X): aucun argument pour max ; -Inf est renvoyé
#> Warning in min(x@data$Y): aucun argument trouvé pour min ; Inf est renvoyé
#> Warning in max(x@data$Y): aucun argument pour max ; -Inf est renvoyé
#> Error in validityMethod(object): invalid extent: xmin >= xmax

# 3 points
las2 = lasfilter(las, Z >= 31.92)
chm <- grid_canopy(las2, res = 0.5, pitfree(thresholds = c(0,2,5,10,15,20)))
#> Error in algorithm(las, layout): Interpolation failed. Input parameters might be wrong.

# 3 points
las2 = lasfilter(las, Z >= 31.92)
chm <- grid_canopy(las2, res = 0.5, pitfree(thresholds = c(0,2,5,10,15,20), subcircle = 0.2))
#> Error in `[.data.table`(dt, , f(X, Y, Z, px, py), by = rownames(dt)): 'by' appears to evaluate to column names but isn't c() or key(). Use by=list(...) if you can. Otherwise, by=eval(rownames(dt)) should work. This is for efficiency so data.table can detect which columns are needed.

# 4 points
las2 = lasfilter(las, Z >= 31.9)
chm <- grid_canopy(las2, res = 0.5, pitfree(thresholds = c(0,2,5,10,15,20)))
#> Error in algorithm(las, layout): Interpolation failed. Input parameters might be wrong.

# 4 points (works)
las2 = lasfilter(las, Z >= 31.9)
chm <- grid_canopy(las2, res = 0.5, pitfree(thresholds = c(0,2,5,10,15,20), subcircle = 0.2))
```

<sup>Created on 2019-10-09 by the [reprex package](https://reprex.tidyverse.org) (v0.3.0)</sup>

mmacander, 2019-10-12:
I am running 2.1.4 and still having an issue with pitfree causing an apply_catalog job to crash.

Jean-Romain, 2019-10-12:
If you are referring to [this issue](https://gis.stackexchange.com/questions/338225/chm-pitfree-error-not-a-matrix) is has not been fixed yet. I'm not able to reproduce. If you have some bug please report in a new issue with a reproducible example. I'm waiting  for a bug report from the person who asked this question on stackexchange (is it you?) before to release 2.2.4.

mmacander, 2019-10-14:
I did not ask the original question but I did just submit a bug report with reproducible example (#288)

commit 6e7272a3 (2019-10-10): Fix #286