r-lidar/lidR #287: tree species classification

hippieschneckeontour, 2019-10-12:
hello, my teacher said that it would be possible to classify tree species with this package (without training data, just with the height and crown shape). Are the segmented trees classified? The result looks like the colors are random. Couldn't find someting about that topic. Would be awesome if somebody could help me!
Thanks.

bi0m3trics, 2019-10-12:
Your teacher, overstated, the possibility of classifying tree species with airborne lidar alone. Of course, this depends on the system your working in and how different individual trees species are (e.g., if one species is short and one is tall or if one an excurrent crown and one has a decurrent crown, then you could build a classify model) but I doubt it I will be trivial. 
In short, your teacher is wrong and there is no "tree species classification" algorithm in lidR. I have a graduate student working on it, but in our system (southwestern conifer-dominated forests), there isn't much detectable difference in species crown form and we're not hoping for great results.

Jean-Romain, 2019-10-12:
There is no tree species classification in the package. Sorry. For a more detailed answer please ask your questions on https://gis.stackexchange.com/ with the tag `lidr`. Github is used to report bug. Question of interest are answered on https://gis.stackexchange.com/ to be useful for other users.

closed 2019-10-12 by Jean-Romain