r-lidar/lidR #288: grid_canopy crash during catalog_apply

mmacander, 2019-10-14:
It returns an error "Not a matrix." I also re-tried after setting opt_stop_early to FALSE (it would be acceptable if occasional tiles were skipped as long as the job continued) but the catalog_apply still crashed, this time with error "in tSearch(D, P, X, threads) : Not a matrix."

`
library(lidR)

dir.create('\~/chm_error')
setwd('\~/chm_error')

download.file("https://seafile.abrinc.com/d/3e2697a0716e4a29b5dd/files/?p=/03VWK140480.las&dl=1",
              "03VWK140480.las", 
              mode="wb")
download.file("https://seafile.abrinc.com/d/3e2697a0716e4a29b5dd/files/?p=/03VWK140490.las&dl=1",
              "03VWK140490.las", 
              mode="wb")
download.file("https://seafile.abrinc.com/d/3e2697a0716e4a29b5dd/files/?p=/03VWK140500.las&dl=1",
              "03VWK140500.las", 
              mode="wb")
download.file("https://seafile.abrinc.com/d/3e2697a0716e4a29b5dd/files/?p=/emmonak_lidar_2016_be.tif&dl=1",
              "emmonak_lidar_2016_be.tif", 
              mode="wb")

ctg <- readLAScatalog(c("03VWK140480.las", "03VWK140490.las", "03VWK140500.las"))
lidR:::catalog_laxindex(ctg)
lascheck(ctg)
plot(ctg, mapview=TRUE, map.type="Esri.WorldImagery")

opt_output_files(ctg) <- "{ORIGINALFILENAME}_normalized"
opt_chunk_buffer(ctg) <- 50
opt_chunk_size(ctg) <- 0

dtm <- raster("emmonak_lidar_2016_be.tif")

#Normalize LAS
catalog_apply(lasnormalize(ctg, dtm, na.rm=TRUE))

ctgn <- readLAScatalog(c("03VWK140480_normalized.las", 
                         "03VWK140490_normalized.las", 
                         "03VWK140500_normalized.las"))

lidR:::catalog_laxindex(ctgn)
lascheck(ctgn)

opt_output_files(ctgn) <- "{ORIGINALFILENAME}_chm_2m"
opt_chunk_buffer(ctgn) <- 50
opt_chunk_size(ctgn) <- 0

thr <- c(0,2,5,10,15)
edg <- c(0, 1.5)
subcircle <- 0.2
res <- 2

catalog_apply(grid_canopy(ctgn, res, pitfree(thr, edg, subcircle)))

#Try again after setting stop_early to FALSE
opt_stop_early(ctgn) <- FALSE
rasterOptions(overwrite=TRUE, datatype="FLT4S")
catalog_apply(grid_canopy(ctgn, res, pitfree(thr, edg, subcircle)))
`

Jean-Romain, 2019-10-14:
Thank you for trying to provide a reproducible example. However if there is a bug in `pitfree` related to [this post](https://gis.stackexchange.com/questions/338225/chm-pitfree-error-not-a-matrix) on stackexchange it won't be reproduced with the code above which is incorrect.

mmacander, 2019-10-15:
Thanks for the correction. The same exact bug does happen using the functions as you suggest.

Jean-Romain, 2019-10-15:
It has been fixed. At 5m, there was only one remaining triangle in tile 2. Sadly R have inconstant behavior with matrix subseting. When subsetting a single row in a matrix, the output is a vector instead of a 1 x n matrix. This is why you got an error `not a matrix`.

commit b1726e6b (2019-10-15): Fix #288
closed 2019-10-15 by Jean-Romain