r-lidar/lidR #289: plot doesn't show color for lasmetrics3d objects

TedwardErker, 2019-10-17:
Thank you for creating this package.  It's truly incredible.

I found a small issue with plotting lasmetrics3d objects.  I ran the example from "grid_metrics3d", the part that is not run fails to plot the colors.

```
## Not run:
     
     # Define your own metric function
     myMetrics = function(i)
     {
       ret = list(
          npoints = length(i),
          imean   = mean(i)
        )
     
        return(ret)
     }
     
     voxels = grid_metrics3d(las, ~myMetrics(Intensity), 3)
     
     plot(voxels, color = "imean")
     #etc.
     ## End(Not run)
     
```

When I run the code above I get no error, but I also don't get any color.  When I run with my own function I get this error: ` Error in col2rgb(colors) : invalid color name 'tree' `.

Not a big deal I think I'll be able to find a work around, but wanted to put it up there.  Thanks again for this amazing tool.  I've only just started using it and I'm excited to find all it can do.

Jean-Romain, 2019-10-17:
Could you please make a fully reproducible example. It works for me. You don't have any color because the color palette is broken by outliers. You can use `trim = 100`. The following do works as expected. I will update the doc to add `trim = 100`. However I'm not able to reproduce the error you mentionned.

```r
library(lidR)

LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
las = readLAS(LASfile)

myMetrics = function(i) {
  ret = list(
    npoints = length(i),
    imean   = mean(i)
  )
  
  return(ret)
}

voxels = grid_metrics3d(las, ~myMetrics(Intensity), 3)
plot(voxels, color = "imean", trim = 100)
```

TedwardErker, 2019-10-17:
Thank you.  Setting trim fixed my visualization problem.  I tracked down my error and think it was from my function not returning numeric (it was boolean).

For posterity, here's the reproducible example:

```
library(lidR)
Modes <- function(x) {
  ux <- unique(x)
  tab <- tabulate(match(x, ux))
  return(list(mode = ux[tab == max(tab)]))
}

LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
las = readLAS(LASfile)

# works great
gmclass <- grid_metrics3d(las, ~Modes(Classification), 3)
plot(gmclass, color = "mode", trim = 3)

lp <- lasdetectshape(las, shp_plane(th1 = 15, th2 = 4, k = 20), "Coplanar")
plot(lp, color = "Coplanar")

gmcoplanar <- grid_metrics3d(lp, ~Modes(Coplanar), 3)
# doesn't work
plot(gmcoplanar, color = "mode")

Modes_numeric<- function(x) {
  ux <- unique(x)
  tab <- tabulate(match(x, ux))
  return(list(mode = as.numeric(ux[tab == max(tab)])))
}

gmcoplanar <- grid_metrics3d(lp, ~Modes_numeric(Coplanar), 3)
# works now
plot(gmcoplanar, color = "mode", trim = 3)
```

Jean-Romain, 2019-10-17:
Ok this is a bug. Boolean data should be displayed. They are displayed for `LAS` objects.

Jean-Romain, 2019-10-17:
TODO

- [ ] add `trim = 100` in the example of the documentation of `grid_metrics3d()`
- [ ] fix `plot.lasmetrics3d()` with boolean

TedwardErker, 2019-10-17:
Great, thanks for addressing this.  Sorry for closing too soon. :)

commit fee9240c (2019-10-17): Fix #289
closed 2019-10-17 by Jean-Romain