r-lidar/lidR #290: LASF 1.2: No ground points found. Impossible to compute a DTM

Leprechault, 2019-10-18:
lidR Community,

I need help with grid_terrain() function, because my output always "No ground points found. Impossible to compute a DTM". I make an inspection in my las file with las_euca and las_euca@data and always sounds OK, despite mu las is LASF 1.2. Please any ideas about why this function don't work for me. Below my code:

library(lidR)
las_euca = readLAS("Euca_27_09_19_dense_cloud.las")
plot(las_euca) # Ok, the image is fine!!

#Digital Terrain Model
dtm= grid_terrain(las_euca, algorithm = kriging(k = 10L))

Error in grid_terrain.LAS(las_euca, algorithm = kriging(k = 10L)) : 
  No ground points found. Impossible to compute a DTM.

las_euca
class        : LAS (LASF v1.2)
point format : 2
memory       : 819.3 Mb 
extent       :-54.561, 343.716, 205.19, 494.044 (xmin, xmax, ymin, ymax)
coord. ref.  : +proj=utm +zone=21 +south +datum=WGS84 +units=m +no_defs +ellps=WGS84 +towgs84=0,0,0 
area         : 102345.5 m²
points       : 10.23 million points
density      : 99.92 points/m²
names        : X Y Z Intensity ReturnNumber NumberOfReturns ScanDirectionFlag EdgeOfFlightline Classification Synthetic_flag Keypoint_flag Withheld_flag ScanAngleRank UserData PointSourceID R G B 

las_euca@data
                X       Y      Z Intensity ReturnNumber NumberOfReturns
       1: -38.954 259.848 42.567         0            1               1
       2: -38.954 259.848 42.567         0            1               1
       3: -38.859 259.750 42.758         0            1               1
       4: -38.859 259.750 42.758         0            1               1
       5: -39.149 259.823 42.474     39578            1               1
      ---                                                              
10226672: 297.344 462.074 76.305     15420            1               1
10226673: 297.252 462.174 76.306      8481            1               1
10226674: 297.376 462.073 76.304     17219            1               1
10226675: 297.252 462.174 76.306      8481            1               1
10226676: 297.312 462.076 76.305     12079            1               1
          ScanDirectionFlag EdgeOfFlightline Classification Synthetic_flag
       1:                 1                0              0          FALSE
       2:                 1                0              0          FALSE
       3:                 1                0              0          FALSE
       4:                 1                0              0          FALSE
       5:                 1                0              0          FALSE
      ---                                                                 
10226672:                 1                0              0          FALSE
10226673:                 1                0              0          FALSE
10226674:                 1                0              0          FALSE
10226675:                 1                0              0          FALSE
10226676:                 1                0              0          FALSE
          Keypoint_flag Withheld_flag ScanAngleRank UserData PointSourceID     R
       1:         FALSE         FALSE             0        0             1     0
       2:         FALSE         FALSE             0        0             1     0
       3:         FALSE         FALSE             0        0             1     0
       4:         FALSE         FALSE             0        0             1     0
       5:         FALSE         FALSE             0        0             1 37779
      ---                                                                       
10226672:         FALSE         FALSE             0        0             1 10537
10226673:         FALSE         FALSE             0        0             1  3341
10226674:         FALSE         FALSE             0        0             1 12336
10226675:         FALSE         FALSE             0        0             1  3341
10226676:         FALSE         FALSE             0        0             1  7710
              G     B
       1:     0     0
       2:     0     0
       3:     0     0
       4:     0     0
       5: 40092 42405
      ---            
10226672: 18504 14392
10226673: 11565  7710
10226674: 20303 16191
10226675: 11565  7710
10226676: 14906 10537

Jean-Romain, 2019-10-18:
Ìf you have any question on how to use the lidR package please ask on https://gis.stackexchange.com/questions/tagged/lidr with the tag `lidr`. This way your question and the answer from the community will be useful for you but also other users. Github is dedicated for bug reports. Thanks.

closed 2019-10-18 by Jean-Romain

khaz666666, 2022-03-03:
Hi,
I have the same issue. I have the latest version of Lidr but still I receive this error. Your help is appreciated!