r-lidar/lidR #291: Error in lmfauto() : could not find function "lmfauto"

LucaRom, 2019-10-23:
Possible small bug. Was testing different things for tree detection and couldn't get lmfauto() to work. 

Here is an example running with the "Example.las" from the Wiki tutorials :

```
library("lidR")
las = readLAS("Example.las")

tops <- tree_detection(las, lmfauto())
```

Would return :

```
> tops <- tree_detection(las, lmfauto())
Error in lmfauto() : could not find function "lmfauto"
```

Jean-Romain, 2019-10-23:
Where have you learned about `lmfauto()`. This is indeed an existing algorithm but is has not been exported to the user. This is something I'm working on and that is in the package for different reason but not for the regular users yet.

Jean-Romain, 2019-10-23:
I think I got it. There is a work-in-progress documentation and this documentation can be found by users despite the fact it should be an internal documentation only. I will remove the documentation. Sorry for the trouble.

LucaRom, 2019-10-23:
No problem, I am not sure where I got the documentation!? Thanks for the follow-up.

commit 01261e5b (2019-10-31): Remove doc of lmfauto close #291
closed 2019-10-31 by Jean-Romain