r-lidar/lidR #293: NAMESPACE: Error in globalVariables(c("mod", "multiple", "name", "type")) : ould not find function "globalVariables"

HenrikBengtsson, 2019-10-26:
Hi, I'm seeing the following R CMD check NOTE:
```
* checking dependencies in R code ... NOTE
Error in globalVariables(c("mod", "multiple", "name", "type")) : 
  could not find function "globalVariables"
```

Using `utils::globalVariables()`, or importing it in NAMESPACE, should fix this.  See R-devel thread '[Rd] missing imports not detected by 'R CMD check' (?)' for a discussion related to this.

Jean-Romain, 2019-10-26:
Hi, could you tell me how to reproduce. I have never used `globalVariables()` ( I `grep`ed it to verify). However, I use `future::getGlobalsAndPackages()`. Especially `c("mod", "multiple", "name", "type")` does not correspond to anything from my code.

HenrikBengtsson, 2019-10-26:
Ah... sorry about the noise.  Searching [GitHub CRAN](https://github.com/search?q=org%3Acran+%22globalVariables%22+AND+%22mod%22+AND+%22multiple%22+AND+%22name%22+AND+%22type%22&type=Code) seems to point to 

https://github.com/cran/manipulateWidget/blob/d42854837e7983d0a61bcb79841cb09ff2ebf3da/R/zzz.R#L88

This comes into play via import path lidR -> rgl -> manipulateWidget.

About reproducing this: See [[Rd] missing imports not detected by 'R CMD check' (?)](https://stat.ethz.ch/pipermail/r-devel/2019-September/078496.html).  It's still open exactly how, but I think it only happens when you install R without the recommended packages.  So, yeah, no easy to reproduce.  Forgetting to import `globalVariables()`, `setClass()`, etc. is quite common (I see it in many many packages) and `R CMD check` does not catch it.  Maybe it will one day, but this far it's only me that caught this problem.  Oh well... I'm closing this one since it's not a lidR bug.

closed 2019-10-26 by HenrikBengtsson