r-lidar/lidR #294: libxt-dev also required on linux ubuntu 19.04

aloboa, 2019-10-28:
We've found that cairo package failed to install:
```
xlib-backend.c:34:10: fatal error: X11/Intrinsic.h: No such file or directory
 #include <X11/Intrinsic.h>      /*->    Xlib.h  Xutil.h Xresource.h .. */
          ^~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [/usr/lib/R/etc/Makeconf:167: xlib-backend.o] Error 1
ERROR: compilation failed for package ‘Cairo’
* removing ‘/home/chris/R/x86_64-pc-linux-gnu-library/3.6/Cairo’

The downloaded source packages are in
	‘/tmp/RtmpP4Gkkj/downloaded_packages’
Warning message:
In install.packages("Cairo") :
  installation of package ‘Cairo’ had non-zero exit status
```

and could solve it by installing libxt-dev (following answer in https://stackoverflow.com/questions/45759518/error-while-attempting-to-install-cairo-r-package-on-centos-7-server)
Just to let you know.

Jean-Romain, 2019-10-28:
Added in the readme, thanks

closed 2019-10-28 by Jean-Romain

nickcrookston, 2023-03-01:
Yup, ran the following and it cleared up the failure I was getting while running install.packages("Cairo")
On Ubuntu...
sudo apt-get install libxt-dev

commit 3a125df8 (2019-10-28): Add libxt-dev (fix #294)