r-lidar/lidR #295: lasnormalize with water?

TedwardErker, 2019-11-03:
This is a small question, maybe a feature request.  It's not very important but I thought I'd put it out there.

I was recently using lasnormalize on a .las file that had points already classified and had a large amount of water.  The algorithm was taking forever (I think because there were no ground points in the area with the water).  I reclassified the water as ground and then ran lasnormalize and it went much faster. 

This works for my application, but should it be done more generally?  I don't know enough to know what is proper from the LAS data specifications, but I would think that many users who want to normalize their points would probably consider "water" to be the same as "ground" (the lowest return which the other points should be normalized against).  Could/should an argument be added to lasnormalize to normalize against ground and water?

Jean-Romain, 2019-11-03:
Could you share a reproducible example? I don't see any obvious reason to explain why it is faster. Also you must specify which algorithm you used. `lasnormalize()` alone does not perform the actual computation.

But in summary your are suggesting to add an argument `water_is_ground = TRUE`. Why not, it is only two lines of code to change. However we must ensure it is relevant in general case and that your case it not a very specific one.

mbedward, 2019-11-03:
I also notice this behaviour with las files for coastal areas where there is a lot of ocean. When I specify the tin algorithm with one of these files, the progress bar text is "Inverse distance weighting" rather than "Searching in TIN"

Jean-Romain, 2019-11-04:
In TIN there is a step of IDW which is necessary to interpolate outside the convex hull of the ground points. Otherwise some point couldn't be normalized. In theory it does no change anything for the output because you are expected to normalize with a buffer. Is is IDW that takes so long?

TedwardErker, 2019-11-04:
I'd be happy to share a reproducible example.  What's the best way for me to share a las file for the reproducible example?  I could provide a link to the ftp from where it came, or upload it somewhere, or whatever you prefer.

I was using the tin() algorithm (sorry for not being explicit).

I think the situation is exactly as you, @Jean-Romain  and @mbedward, are describing.  There are some very large lakes in my catalog of las files.

Jean-Romain, 2019-11-04:
Share a link to one file if it is already on a ftp server + reproducible code to compare runtimes with and without using water as ground. I will deal with that.

TedwardErker, 2019-11-04:
Data over picnic point of lake mendota.  I chose an extreme example.
```
wget ftp://ftp.ssec.wisc.edu/pub/wisconsinview/lidar/Dane/Dane_2017_3DEP_Delivery/Classified_LAS/ESRI/0820_esri.las
```

```
  library(lidR)
  l <- readLAS("/home/erker/hgt_data/reproducibleexample/0820_esri.las")
  plot(l, color = "Classification")

# commented out because so slow.... 
#    system.time(ln_keepwater <- lasnormalize(l, tin()))

# is this the best way to reclassify points?
  l@data$Classification[l@data$Classification == 9] <- 2L

  system.time(ln_removewater <- lasnormalize(l, tin()))

```

Without reclassifying water it was taking a very long time (over 5 minutes then I aborted it).  With reclassifying water it took 24 seconds.

Jean-Romain, 2019-11-04:
It is indeed and extreme case. 

`tin()` makes the triangulation of ground point. In your case few points spread into a small region of 2 km²  among a region of 20 km² (if units are meter because there is no CRS in the file).  ~90% of the points are outside the ground points. Consequently they cannot be interpolated using a triangular network which is undefined outside the ground points. For these points `tin()` uses `knnidw()`. In general case it corresponds to very few points close to actual ground points. In your case it correspond to 90% of the point very far from the ground point. The knn search is consequently slow and is applied to 90% of the points instead of being fast and applied to 0.1% of the points. This is why the computation never ends.

If you change water as ground, then, ground points are spread evenly on the whole coverage, this is the regular case and it is fast (24 second on my computer as well).

In lidR v2.2.0 I added an argument `add_class` in `lasnormalize()`.

```r
library(lidR)
l <- readLAS("0820_esri.las", select = "c")
ln <- lasnormalize(l, tin(), add_class = 9))
```

TedwardErker, 2019-11-04:
You are outstanding.

(units are feet: epsg:7599).

closed 2019-11-05 by Jean-Romain

mbedward, 2019-11-05:
This is working very well for my coastal LAS files. Many thanks!

Jean-Romain, 2019-11-05:
Great! Remember that the development version may change. I can change the name of the parameter, I can chose to modify the default behavior of `add_class` to be 2 instead of `NULL` or whatever. The feature will stay but the front end may change. Keep an eye to the changelog if you update the package again.

mbedward, 2019-11-05:
Understood - and that's all fine. Thanks again!

Jean-Romain, 2019-11-11:
@mbedward @TedwardErker 

`grid_terrain` and `lasnormalize` now have a param `use_class = c(2L, 9L)`. I removed `add_class` and I put the water by default. This way even the class 2 can be changed by the user `use_class = 123L`).

mbedward, 2019-11-13:
That's ideal - thanks Jean-Romain