r-lidar/lidR #296: grid_metrics3d not working for low vertical resolutions

Senn-J, 2019-11-05:
Hi,
thanks for developing this extremely useful package.
I am experiencing an issue when using grid_metrics3d for low vertical resolutions. This happens both for cubic and non-cubic voxels. The function produces empty output for` res = 1` or `res = c(5,1)` whereas `res = 5` or `res = c(5,6)` produces actual values.
`
metr <- grid_metrics3d(las, ~LAD(Z, dz=1, k=0.5, z0=0), res =1)`

Is this a known issue or am I just overlooking something.

cheers,
Johannes

Jean-Romain, 2019-11-05:
Hi,

When reporting something please provide a reproducible example including a small dataset (use one included in the package if possible). But here the problem seems clear. Using `LAD` in `grid_metrics` does not make sense. Indeed the output of `LAD` is not a number. In `grid_metrics3d` it has even less sense because `LAD` does not have a significance at this scale. For a more detailed answer please ask on https://gis.stackexchange.com/. I usually provide detailed answers to the benefit of the community.

closed 2019-11-05 by Jean-Romain

Senn-J, 2023-02-02:
Hi Jean-Romain,
thank you for developing this extremely useful and efficient package, I really appreciate that. And also thank you for citing me even though what I did 
> does not make sense.
In the meantime I had a tweak using the voxel_metrics function with incredibly tall voxels. Now coming back to my old code I am experiencing same old problem again. 
is there a way to calculate the gap fraction profile automatically for each pixel?  i keep getting errors and there seems to be no way to return a list or data.frame to any of the *_metrics functions. this would be extremely useful. 
Thank you
Johannes