r-lidar/lidR #298: Use some tiles as "buffer only" when processing a LAScatalog

Jean-Romain, 2019-11-13:
This feature request comes from  [stackexchange](https://gis.stackexchange.com/questions/341631/ignoring-buffer-files-in-processing). When processing a LAScatalog it might be suitable to process only a subset of the LAScatalog but keep the other tiles to ensure to load a buffer everywhere.

Jean-Romain, 2019-11-13:
lidR 2.2.0 now have this capability. See examples below. As usual, it is the development  version so it might change a bit before the actual release.

``` r
library(lidR)
ctg = readLAScatalog("/mydata/")

# Add an attribute 'processed' to tell which file to actually process
processed <- rep(FALSE, nrow(ctg))
processed[c(7:9, 12:14)] <- TRUE
ctg$processed <- processed

# plot function has been updated to show the files processed/not processed
plot(ctg)
```

[image]

``` r
# Plot to see if the chunk are correctly computed
opt_chunk_size(ctg) <- 0
plot(ctg, chunk = TRUE)
```

[image]

``` r
opt_chunk_size(ctg) <- 500
plot(ctg, chunk = TRUE)
```

[image]

``` r
opt_chunk_size(ctg) <- 700
plot(ctg, chunk = TRUE)
```

[image]

``` r
opt_chunk_size(ctg) <- 500
opt_chunk_alignment(ctg) <- c(75, 75)
plot(ctg, chunk = TRUE)
```

[image]

``` r
# Because chunks are necessarily rectangular there are some 'imperfect cases'
processed <- rep(FALSE, nrow(ctg))
processed[c(7:9, 13L, 18L)] <- TRUE
ctg$processed <- processed
plot(ctg)
```

[image]

``` r
opt_chunk_size(ctg) <- 500
plot(ctg, chunk = TRUE)
```

[image]

``` r
#  Try to process
opt_chunk_size(ctg) <- 0
m = grid_metrics(ctg, ~mean(Z), 20)
plot(m, col = height.colors(50))
```

[image]

The function `catalog_select()` has also been updated to flag some tiles.

```r
ctg <- catalog_select(ctg, method = "flag_processed")
```

lucas-johnson, 2019-11-13:
Awesome, thanks for implementing this.

lucas-johnson, 2019-11-13:
Do you have any estimate when 2.2 will be released?

Jean-Romain, 2019-11-14:
You can already install 2.2 from github with

```r
remotes::install_github("Jean-Romain/lidR", ref = "devel")
```

but I plan to release hopefully on Jan 1st 2020.

closed 2019-11-14 by Jean-Romain