r-lidar/lidR #302: Bug in "buffer only" file processing

lucas-johnson, 2019-12-03:
It looks like there is a bug in the functionality implemented here https://github.com/Jean-Romain/lidR/issues/298

I want to normalize a subset of a catalog and use the remaining files as buffers for the subset. 
I have altered the `processed` flag on the catalog so that my catalog looks like this: 

[image]

It is clear that I have designated 5 files to process:
```
ctg[ctg$processed, ]$filename
[1] "/tmp/download/u_5505089900_2015.las" "/tmp/download/u_5505090050_2015.las" "/tmp/download/u_5505090200_2015.las"
[4] "/tmp/download/u_5520089900_2015.las" "/tmp/download/u_5520090050_2015.las"

```

And the remaining 7 should be used for buffering only:
```
ctg[!ctg$processed, ]$filename
[1] "/tmp/download/u_5490089750_2015.las" "/tmp/download/u_5490089900_2015.las" "/tmp/download/u_5490090050_2015.las"
[4] "/tmp/download/u_5490090200_2015.las" "/tmp/download/u_5505089750_2015.las" "/tmp/download/u_5520089750_2015.las"
[7] "/tmp/download/u_5535089750_2015.las"
```

The results of normalizing the catalog seem good at a glance:
[image]

However, after digging deeper it appears that 5 of the 7 "buffer only" files have actually been processed (compare the filenames to buffer only filenames above):
```
normalized$filename
[1] "/tmp/normalized//u_5490089750_2015.las" "/tmp/normalized//u_5490089900_2015.las"
[3] "/tmp/normalized//u_5490090050_2015.las" "/tmp/normalized//u_5490090200_2015.las"
[5] "/tmp/normalized//u_5505089750_2015.las"
```

Please let me know if you need more information from me. If I understand the functionality of "buffer only" described here: https://github.com/Jean-Romain/lidR/issues/298 then this should be a bug. Perhaps I have misunderstood how the functionality should be applied...

lucas-johnson, 2019-12-03:
From looking at the plots it actually seems the right file was processed, however they might have been given incorrect names when writing them with the option: `opt_output_file(ctg) = {ORIGINALFILENAME}`

Jean-Romain, 2019-12-03:
Thank you for reporting. I did not tested it a lot after i made it. Actually I was counting on you to test it :wink:. I think you spotted the right problem. The names are badly attributed. That makes a lot of sense actually. I will fix it quickly.

Jean-Romain, 2019-12-03:
I think I fixed it in https://github.com/Jean-Romain/lidR/commit/c61e4f2f2948183f81021a9e3f4dae58fc5d2f97 but I didn't tested it yet. But I'm pretty sure it is ok. Please let me know.

lucas-johnson, 2019-12-03:
Thank you so much for getting to this so quickly! I'll test it tonight and let you know if it produces the desired results.

lucas-johnson, 2019-12-04:
Looks like it's working now and the filenames are correctly attributed. Thanks again.

closed 2019-12-04 by Jean-Romain