r-lidar/lidR #303: grid_metrics() not finding function when run in parallel

Myxomycota, 2019-12-06:
I'm having an interesting issue using the grid_metrics function within a parLapply. The issue is related to writing a custom function, and handing it off to grid_metrics to be applied to a LAS file.
The function I'm trying to apply is defined as: 
`getQuants <- function(x,i){
  q25 <- function(x){quantile(x,.25)}
  
  q50 <- function(x){quantile(x,.5)}
  
  q75 <- function(x){quantile(x,.75)}
  
  qMax <- function(x){max(x)}
  
  q25i <- function(x,i){
    q25=quantile(x,.25)
    mean(i[x<q25 & x>0],na.rm=TRUE)
  }
  
  q50i <- function(x,i){
    q25=quantile(x,.25)
    q50=quantile(x,.50)
    mean(i[x<q50 & x>q25],na.rm=TRUE)
  }
  
  q75i <- function(x,i){
    q50=quantile(x,.5)
    q75=quantile(x,.750)
    mean(i[x<q75 & x>q50],na.rm=TRUE)
  }
  
  qMaxi <- function(x,i){
    q75=quantile(x,.750)
    mean(i[x>q75],na.rm=TRUE)
  }
  
  list(q25 = q25(x),
       q50 = q50(x),
       q75 = q75(x),
       qMax = qMax(x),
       q25i = q25i(x,i),
       q50i = q50i(x,i),
       q75i = q75i(x,i),
       qMaxi = qMaxi(x,i))
}`

When I run the function in an lapply, I have no issues; I make sure to define the function internally to the function defined in either the parLapply or lapply. When I run the function in an lapply I have no issues. When I run the function in a parLapply over non-LAS data (for example:
`cl <- makeCluster(2)
1:4 %>% parLapply(cl,.,function(x,getQuants){
  library(data.table)
  dt.examp <- data.table(A=runif(100,1,100),B=runif(100,1,100))
  
  getQuants(dt.examp$A,dt.examp$B)
  },getQuants)
stopCluster(cl)`
), the function runs without issue. 

I think this is some kind of environments issue and perhaps related to how lidR expects to run in parallel, but I'm really not sure how to resolve it.

Thanks.

Jean-Romain, 2019-12-06:
Do not use custom `*apply` function especially with parallel backend. Use the LAScatalog processing engine. See:

```r
?lidR::`LAScatalog-class`
?lidR::catalog_apply
?lidR::`lidR-parallelism`
```

See also the vignettes

https://cran.r-project.org/web/packages/lidR/vignettes/lidR-LAScatalog-class.html
https://cran.r-project.org/web/packages/lidR/vignettes/lidR-catalog-apply-examples.html

Myxomycota, 2019-12-06:
Thanks Jean-Romain, this appears to have resolved the issue.

closed 2019-12-06 by Myxomycota