r-lidar/lidR #304: [2.1.5] lastransform doesn't update scale

spono, 2019-12-09:
Ciao JR,
I have a bunch of files to convert from EPSG 3035 to 32632. When reading them I get a warning:
```
Warning messages:
1: Invalid header: X scale factors should be factor ten of 0.1 or 0.5 or 0.25 not 1e-07 
2: Invalid header: Y scale factors should be factor ten of 0.1 or 0.5 or 0.25 not 1e-07
```
and indeed `las@header` shows:
`Scale factor X Y Z:       1e-07 1e-07 0.001 `

The transformation seems to work fine (coords are ok) but then when I try to write the las I get the related error:
`Errore: Invalid header: X scale factors should be factor ten of 0.1 or 0.5 or 0.25 not 1e-07`

Is it something that the user should fix manually or what? I don't know if might be similar to #270 .
Thanks in advance!

Jean-Romain, 2019-12-09:
You have a precision of 10e-7 m in your data. It means that your points are located with a precision of a tenth of a micrometer (the size of a virus). This is actually valid and it should be writable in a LAS file but I hard coded a limited number of valid scale factors until 1e-4 (100 micrometers). Use `lasrescale` to set a reasonable scale factor.

Jean-Romain, 2019-12-09:
Also I really advice you to use LAStools for this kind of processing.

spono, 2019-12-09:
> 
> 
> You have a precision of 10e-7 m in your data. It means that your points are located with a precision of a tenth of a micrometer (the size of a virus). This is actually valid and it should be writable in a LAS file but I hard coded a limited number of valid scale factors up-to 1e-4 (100 micrometers). Use `lasrescale` to set a reasonable scale factor.
> 
> The message should mention that ten of 0.1 means 0.1, 0.01, 0.001, 0.0001 but not 0.0000001

oh, true, at a first glimpse I didn't realise it was that small, I thought it was related to the CRS scale (degrees) and would have been changed accordingly.
[I don't know if I should ask here or in stackexchange, but] how should be identified a "reasonable scale factor" if it is not reported in (meta)data?
(BTW thanks)

Jean-Romain, 2019-12-09:
Well, in degrees seven digits can make sense. It is roughly a centimeter precision. I should maybe allow for such accuracy. Honestly I'm not an expert at this stage. If we are limited to 4 digits we can't store coordinates in degree. In another hand if I do not put strong constraint I can't handle case like your. A millimeter accuracy is way enough. But I think I will allow accuracy up to 10-7 for the case of data in degrees.

Jean-Romain, 2020-02-23:
This issue should be fixed in rlas 1.3.6 https://github.com/Jean-Romain/rlas/commit/00384988c3e3a7845d36b5e29d47c9f593b6d6a3

closed 2020-02-23 by Jean-Romain