r-lidar/lidR #305: Set operations with lidR?

jgrn307, 2019-12-11:
Is it possible to do set operations (given some distance threshold) between two point cloud datasets?  This would be a filter such as intersections, unions, differences, etc.  Related: is there a point-to-point distance tool between two lidar datasets?

Jean-Romain, 2019-12-11:
No there is no such tool. However I'm interested in understanding what you have in mind. How do you define formally the meaning of the difference (and other set operations) of two a point clouds?

jgrn307, 2019-12-11:
So a standard set operation would be, for instance, an intersection -- which given two point clouds A and B, it returns only points that are members of BOTH point clouds (this is not the same as a merge, mind you).  Many of the GEOS functions (https://cran.r-project.org/web/packages/rgeos/rgeos.pdf) I think would be INCREDIBLY useful if ported to three-dimensions (they are all designed for two-dimensions).  The output of these would be a new point cloud based on the selections (and potentially merging of fields).

Personally, I'm actually interested in distance-based selections -- e.g. "all points in A that are within some distance of points in B" which, to begin with, would require the ability to calculate point-to-point distances and then do some logic on those.  The only point-to-point distance calculator I'm aware of is in CloudCompare, which has challenges with command line mode and batch processes.

A really good application for this is multitemporal analyses -- if I want to look for points that represent e.g. a loss, I would look at points that were present in a dataset collected at time 1, and were not present at time 2 (so the asymmetrical difference) -- but this would need a distance threshold since there's very little chance two points will be in EXACTLY the same position.

bi0m3trics, 2019-12-11:
Just thinking, regarding "all points in A that are within some distance of points in B" – isn’t this what I did for step 1 of the snag segmentation code (minus the counting the number of points part) or am I missing something? Then in step 2 go on to calculate the mean branch-bole ratio for intensity, but this could easily be a distance calculation in a sphere of size x… it seems to me that code could just be tweaked/recycled.

Jean-Romain, 2019-12-11:
What you are defining here are the operations for polygons. They cannot be extended to point cloud as is. In a point cloud points the members of A and B are basically the null set. Expect if you consider the concave hull of the point cloud. But concave hulls are not unique.

> but this would need a distance threshold since there's very little chance two points will be in EXACTLY the same position.

Yep, so you are defining the intersection as the points of A that have a nearest neighbor closer than a threshold in B. Why not. Is there a consensus on this definition?

Jean-Romain, 2019-12-11:
> isn’t this what I did for step 1 of the snag segmentation code (minus the counting the number of points part) or am I missing something?

I don't remember but in my memories the snags segmentation relies on regular queries. But we can look in the sources to check. But yes we have the internal tools to make something like that.

jgrn307, 2019-12-12:
Re: polygons -- intersections without a distance threshold are pretty uncommon -- basically its a duplicate point detector (which might be useful in certain cases, e.g. if you've created point fields in different processes and want to merge them).  The core algorithm I think I'm asking for is 1) the ability to calculate point to nearest point distances between two point clouds (assigning this to a new field in the x,y,z file) and allowing for filtering on this (filtering is already part of your workflow).  It looks like Andrew above may have already done the basic work for this -- is this something that you could tweak to provide minimally a point-to-point distances:

https://www.cloudcompare.org/doc/wiki/index.php?title=Cloud-to-Cloud_Distance

I think the trick with this is not blowing up memory and using some octree-style approach to the calculations -- the memory will scale by the product of the number of points in each cloud without something like that (since the brute force method will be to compare each point with each other point).

jgrn307, 2019-12-12:
Here's the brute force approach I'm thinking:

`
# Two point clouds:
cloudA = matrix(c(runif(4),runif(4),runif(4)),ncol=3)
cloudB = matrix(c(runif(10),runif(10),runif(10)),ncol=3)

colnames(cloudA)=colnames(cloudB)=c("x","y","z")
# Calculate coordinate distances of A to B:

x_difference <- outer(cloudA[,"x"],cloudB[,"x"],"-")
y_difference <- outer(cloudA[,"y"],cloudB[,"y"],"-")
z_difference <- outer(cloudA[,"z"],cloudB[,"z"],"-")

difference_matrix <- sqrt(x_difference^2+y_difference^2+z_difference^2)

A_to_B_distance <- apply(difference_matrix,1,min)
B_to_A_distance <- apply(difference_matrix,2,min)`

Jean-Romain, 2019-12-12:
We have all the internals to implement that efficiently in ALS data. I retain the suggestion.

Jean-Romain, 2019-12-12:
But this won't be efficient for other point cloud. `lidR` is designed for ALS. It relies on a fast 2D spatial indexation we don't have an octree.

jgrn307, 2019-12-12:
This particular problem could easily be done in "chunks" to help with memory and maybe a nested loop -- read X points from cloud A, and Y points from cloud B (where X and Y are chosen to keep the memory down), do the calculations, preserve the "current min" for the "A" chunk, go to the next set in B, etc, etc., then move to the next chunk of A, and repeat.  Larger chunk sizes would mean less repeated reading (which would occur in the B point cloud in this implementation).

Jean-Romain, 2019-12-12:
You know that you can use `RANN` or `FNN` to search knn and get the distance.

```r
RANN::nn2(ref, query, k = 1)
```

I can make a faster and less memory greedy version of it but this is already a good solution.

jgrn307, 2019-12-12:
Cool -- I'll try to implement those in the short term, but if you come up with a memory safe version of those that would be great!

Jean-Romain, 2019-12-12:
You request is definitively a good request that I will consider but it is not a priority. I'll keep you informed.

jgrn307, 2019-12-13:
So the function seems pretty straightforward given your suggestions:

```
c2c_dist <- function(las1,las2)
{
	las1_coords <- data.frame(X=las1@data$X,Y=las1@data$Y,Z=las1@data$Z)
	las2_coords <- data.frame(X=las2@data$X,Y=las2@data$Y,Z=las2@data$Z)
	
	c2c_distance <- c(RANN::nn2(las2_coords,las1_coords,k=1)$nn.dists)
	las1_distance <- lasaddextrabytes(las1,c2c_distance,name="c2c_dist",desc="C2C absolute distances")
	return(las1_distance)
}
```

I guess the one thing I'm curious about is how you would implement this as a lidR catalog function, given it really needs TWO catalogs thrown at it.

closed 2021-03-18 by Jean-Romain