r-lidar/lidR #306: [2.1.5] lasmergespatial fails for non-in-memory RasterStack

spono, 2019-12-12:
I want to colorise a point cloud with a `RasterStack`

```
las = readLAS('clip_2019.laz', select='xyz')
crs(las) = sp::CRS('+init=epsg:32632')

r = stack('orthoimage.tif')
crs(r) = sp::CRS('+init=epsg:32632')

las = lasmergespatial(las, r)
```

but I get the following warning:
```
Warning messages:
1: In max(R, na.rm = TRUE) :
  no non-missing arguments to max; returning -Inf
2: In max(G, na.rm = TRUE) :
  no non-missing arguments to max; returning -Inf
3: In max(B, na.rm = TRUE) :
  no non-missing arguments to max; returning -Inf
```

The RGB columns are added to `las` but they contain only `NAs`.
```
> head(las)
          X       Y        Z  R  G  B
1: 729978.2 5149051 1512.568 NA NA NA
2: 729978.5 5149051 1519.967 NA NA NA
3: 729979.0 5149051 1512.729 NA NA NA
4: 729979.0 5149051 1512.828 NA NA NA
5: 729979.1 5149051 1512.847 NA NA NA
6: 729979.3 5149051 1512.745 NA NA NA
```

Everything works fine when I force the load with `readAll`, as proposed in the post mentioned in bug report #285 .
```
> head(las)
          X       Y        Z     R     G     B
1: 729978.2 5149051 1512.568 30069 31868 22616
2: 729978.5 5149051 1519.967 30326 32125 23130
3: 729979.0 5149051 1512.729 29555 31354 22359
4: 729979.0 5149051 1512.828 29298 31354 22102
5: 729979.1 5149051 1512.847 28527 30840 21588
6: 729979.3 5149051 1512.745 27499 30326 21074
```

Jean-Romain, 2019-12-12:
Fixed in 2.2.0 because 2.1.5 won't be released.

Jean-Romain, 2019-12-12:
Fixed also in 2.1.5 btw

commit 0067225d (2019-12-12): Fix #306
commit f5abd06d (2019-12-12): Fix #306
closed 2019-12-12 by Jean-Romain