r-lidar/lidR #308: opt_stop_early not working properly

lucas-johnson, 2019-12-13:
### Current behavior
The `opt_stop_early` setting doesn't effect the `LAScatalog` processing:
```
> opt_stop_early(testcat) = FALSE
> lasnormalize(testcat, knnidw(k=5))
Processing [==============================================>-----------------------------------------------]  50% (1/2) eta:  1sError in lasnormalize.LAS(x, algorithm, na.rm, use_class, ...) : 
  No ground point found in the point cloud.
> opt_stop_early(testcat) = TRUE
> lasnormalize(testcat, knnidw(k=5))
Processing [==============================================>-----------------------------------------------]  50% (1/2) eta:  1sError: No ground point found in the point cloud.
> 
```

### Desired behavior
When `opt_stop_early(ctg)` is set to `FALSE` then the processing of the catalog should continue despite encountering an error with one of the chunks.

Jean-Romain, 2019-12-13:
Reproducible example

```r
library(lidR)
ctg = readLAScatalog("~/Documents/ALS data/Haliburton dataset/Landscape LiDAR/")

lidR:::catalog_fakerun(ctg, pwarning = 0, perror = 0.1)

opt_stop_early(ctg) <- FALSE
lidR:::catalog_fakerun(ctg, pwarning = 0, perror = 0.1)
````

Jean-Romain, 2019-12-13:
Fixed in v2.1.5

closed 2019-12-13 by Jean-Romain