r-lidar/lidR #310: `is.overlapping(ctg)` is too sensitive

lucas-johnson, 2020-01-04:
The is.overlapping flag seems too sensitive. In the following case, the total area of overlap in the catalog is less than 1 square mm. It might be better to set a threshold greater than 0, only over which the `is.overlapping()` function would return `TRUE`. I'm not sure what that threshold should be (.01% of total area, or 1 square unit of area), but the current threshold seems too sensitive. 
 
Example:
```
> is.overlapping(ctg)
[1] TRUE
> spdf <- as.spatial(ctg)
> contour <- rgeos::gUnaryUnion(spdf)
> actual_area <- round(contour@polygons[[1]]@area, 4)
> measured_area <- round(area(ctg), 4)
> actual_area < measured_area
[1] TRUE
> actual_area
[1] 54000000
> measured_area
[1] 54000000
> measured_area - actual_area
[1] 0.0001000017
> ctg
class       : LAScatalog
extent      : 366000 , 372000 , 4717500 , 4726500 (xmin, xmax, ymin, ymax)
coord. ref. : +proj=utm +zone=18 +ellps=GRS80 +units=m +vunits=m +no_defs 
area        : 54 km²
points      : 311.79 million points
density     : 5.8 points/m²
num. files  : 24 
```

Jean-Romain, 2020-01-04:
You are right this should be improve. 

1 square unit threshold seems a reasonable option but I think it will fail if the points are stored in degrees. A percentage is maybe not the best option because it won't detect a duplicated file in a large dataset.

I could also add an argument in `is.overlapping` print/plot a report and show where are the overlaps and how much it is overlapping.

Could you update your issue by updating the values without `round(..., 4)` to see the actual values.

Jean-Romain, 2020-01-04:
What do you think of a new function `overlaps()`

``` r
overlaps <- function(catalog) {
  spdf <- as.spatial(catalog)
  ii <- rgeos::gIntersects(spdf, byid = TRUE, returnDense = FALSE)
  
  intersections <- vector("list", length(ii))
  
  for (i in 1:length(ii)) {
    k <- ii[[i]]
    k <- k[k != i]
    p <- spdf[k, ]
    q <- spdf[i, ]
    w <- rgeos::gIntersection(q, p)
    w@polygons[[1]]@ID <- as.character(i)
    intersections[[i]] <- w
  }
  
  output <- do.call(rbind, intersections)
  output <- rgeos::gUnaryUnion(output)
  return(output)
}

library(lidR)
#> Le chargement a nécessité le package : raster
#> Le chargement a nécessité le package : sp
par(mfrow=c(1,2))
ctg <- readLAScatalog("~/Documents/ALS data/BCTS/")
#> Be careful, some tiles seem to overlap each other. lidR may return incorrect outputs with edge artifacts when processing this catalog.
over <- overlaps(ctg)

plot(ctg)
plot(over, add = T, col = "red", border="red", lwd = 0.5)

plot(over, add = F, col = "red", border = "red", lwd = 0.5)
```

[image]

<sup>Created on 2020-01-04 by the [reprex package](https://reprex.tidyverse.org) (v0.3.0)</sup>

lucas-johnson, 2020-01-04:
I have updated my post to display the unrounded areas...

I like the `overlaps(ctg)` function. This would be very helpful for me.

Jean-Romain, 2020-01-04:
Please share your `LAScatalog` using `saveRDS()` or `dput()`

lucas-johnson, 2020-01-04:
I'm not sure how you want me to share the catalog. It is quite large. I tried compressing the tiles to `laz` files and zipping them, but the resulting zip file is 1.24 GB which is too big to post here.

lucas-johnson, 2020-01-04:
[test_ctg.txt](https://github.com/Jean-Romain/lidR/files/4022176/test_ctg.txt)

Jean-Romain, 2020-01-04:
I don't need your point cloud. I only need the R object that represents the `LAScatalog` which is only few KB. Using `saveRDS` you can write any R object in a binary file. Using `dput` you generate a copy/pastable string that will allow me to reconstruct the R object. Both are correct.

lucas-johnson, 2020-01-04:
Ok understood. I used `dput` to save the catalog to a `.txt` file:

[test_ctg.txt](https://github.com/Jean-Romain/lidR/files/4022197/test_ctg.txt)

Jean-Romain, 2020-01-04:
Well, actually a `LAScatalog` is a complex object and its constructor is not properly written. `dput` does not enable me to reconstrut the object. Please use `saveRDS`.

lucas-johnson, 2020-01-04:
Github doesn't support the `.rds` extension, so I have saved the RDS object as a `.txt` file:
[test_ctg.txt](https://github.com/Jean-Romain/lidR/files/4022221/test_ctg.txt)

Jean-Romain, 2020-01-04:
Ok I understood your issue. The header is not (strictly) well populated. It gives the extent of the files instead of the the coordinates of the most right/left/top/bottom points. For example `Max X` on the left tile is 367500 and `Min X` on the right is 367500 as well. So the intersection area is 0 and the intersection shape is a line.

It is expected, for example, that `Max X` on the left tile is 367499.99 and `Min X` on the right is 367500.02.

Anyway this should give an intersection area of 0 but we get 0.0001000017 which is likely to be an precision error due to how computers work.

Jean-Romain, 2020-01-04:
I edited the function `overlaps` above. I will fix that in `is.overlapping`.

lucas-johnson, 2020-01-04:
I'm seeing another edge case in `overlaps` where:
```
w <- rgeos::gIntersection(q, p)
```
results in a `SpatialCollections` object containing 2 lines and 1 point. 

w:
```
w
class       : SpatialCollections 
features    : 1 
extent      : 363011.3, 364500, 4720500, 4722000  (xmin, xmax, ymin, ymax)
crs         : +proj=utm +zone=18 +ellps=GRS80 +units=m +vunits=m +no_defs 
```

p:
```
plot(p)
```
[image]

q:
```
plot(q, add=TRUE, col="red")
```
[image]

Overlaps:
```
plot(w@lineobj, col="blue")
plot(w@pointobj, col="blue", add=TRUE)
```
[image]

Error Message:
```
 Error in overlaps(catalog) : 
  no slot of name "polygons" for this object of class "SpatialCollections" 
```

Problematic catalog: 
[test2_ctg.txt](https://github.com/Jean-Romain/lidR/files/4022379/test2_ctg.txt)

Jean-Romain, 2020-01-04:
This one handle this edge case

```r
overlaps <- function(catalog) 
{
  spdf <- as.spatial(catalog)
  ii <- rgeos::gIntersects(spdf, byid = TRUE, returnDense = FALSE)
  
  intersections <- vector("list", length(ii))
  
  for (i in 1:length(ii)) 
  {
    k <- ii[[i]]
    k <- k[k != i]
    
    if (length(k) == 0) next
    
    p <- spdf[k, ]
    q <- spdf[i, ]
    w <- rgeos::gIntersection(q, p)
    
    if (is(w, "SpatialLines")) next
    
    if (is(w, "SpatialCollections")) {
      if (is.null(w@polyobj)) next
      w <- w@polyobj
    }
    
    w@polygons[[1]]@ID <- as.character(i)
    intersections[[i]] <- w
  }
  
  if (all(sapply(intersections, is.null))) {
    empty <- SpatialPolygons(list())
    raster::projection(empty) <- raster::projection(catalog)
    return(empty)
  }
  
  output <- do.call(rbind, intersections)
  output <- rgeos::gUnaryUnion(output)
  return(output)
}
```

Jean-Romain, 2020-01-04:
In `v2.2.0` (release on jan 6th) I added a small tolerance.
In `v2.3.0` I will add the function `overlaps`

commit a678a7d0 (2020-03-17): Add overlaps = TRUE (close #310)
closed 2020-03-19 by Jean-Romain