r-lidar/lidR #311: Remove dependence to gdalUtils

Jean-Romain, 2020-01-11:
Message from CRAN below. `gdalUtils` dependency must be urgently removed and will generate a deep backward incompatibility.

> This concerns the CRAN packages
> 
>   MODISSnow MODIStsp gdalUtils lidR nhdR plainview raptr reproducible
>   uavRmp
> 
> maintained by one of you:
> 
>   Chris Reudenbach <reudenbach@uni-marburg.de>: uavRmp
>   Eliot J B McIntire <eliot.mcintire@canada.ca>: reproducible
>   Jean-Romain Roussel <jean-romain.roussel.1@ulaval.ca>: lidR
>   Jeffrey O Hanson <jeffrey.hanson@uqconnect.edu.au>: raptr
>   Johannes Signer <jmsigner@gmail.com>: MODISSnow
>   Jonathan Asher Greenberg <gdalUtils@estarcion.net>: gdalUtils
>   Joseph Stachelek <stachel2@msu.edu>: nhdR
>   Lorenzo Busetto <lbusett@gmail.com>: MODIStsp
>   Tim Appelhans <tim.appelhans@gmail.com>: plainview
> 
> We have repeatedly asked for an update fixing the check problems
> shown on
>   <https://cran.r-project.org/web/checks/check_results_gdalUtils.html>
> with no reply from the maintainer thus far.
> 
> Thus, package gdalUtils is now scheduled for archival on 2020-01-25,
> and archiving this will necessitate also archiving its strong reverse
> dependencies.
> 
> Please negotiate the necessary actions.

jgrn307, 2020-01-11:
I'll try to get this fixed this weekend.

J

Chris Reudenbach reudenbach@uni-marburg.de<mailto:reudenbach@uni-marburg.de>: uavRmp
Eliot J B McIntire eliot.mcintire@canada.ca<mailto:eliot.mcintire@canada.ca>: reproducible
Jean-Romain Roussel jean-romain.roussel.1@ulaval.ca<mailto:jean-romain.roussel.1@ulaval.ca>: lidR
Jeffrey O Hanson jeffrey.hanson@uqconnect.edu.au<mailto:jeffrey.hanson@uqconnect.edu.au>: raptr
Johannes Signer jmsigner@gmail.com<mailto:jmsigner@gmail.com>: MODISSnow
Jonathan Asher Greenberg gdalUtils@estarcion.net<mailto:gdalUtils@estarcion.net>: gdalUtils
Joseph Stachelek stachel2@msu.edu<mailto:stachel2@msu.edu>: nhdR
Lorenzo Busetto lbusett@gmail.com<mailto:lbusett@gmail.com>: MODIStsp
Tim Appelhans tim.appelhans@gmail.com<mailto:tim.appelhans@gmail.com>: plainview

https://cran.r-project.org/web/checks/check_results_gdalUtils.html<https://nam04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fcran.r-project.org%2Fweb%2Fchecks%2Fcheck_results_gdalUtils.html&data=01%7C01%7Cjgreenberg%40unr.edu%7C412ab3ac797c4a7ec6cf08d796780d4a%7C523b4bfc0ebd4c03b2b96f6a17fd31d8%7C1&sdata=9EJ5vNdhfZcdogaj4Ck052VyIIvuhUHy9KPJ34DjYI8%3D&reserved=0>

Jean-Romain, 2020-01-11:
Great thank you

Jean-Romain, 2020-01-16:
`gdalUtils` is back on CRAN

closed 2020-01-16 by Jean-Romain

jgrn307, 2020-01-16:
Sorry for the panic :)  The change to how class(matrix) behaves broke a bunch of things in gdalUtils, all of which are now fixed!